meshtastic/Meshtastic-Apple #526: 🐞 [Bug]: Channel history getting mixed up when Channel order is changed

AssetBurned, 2024-02-25:
### Firmware Version

2.2.22.404d0dd beta

### What did you do?

1) Setup the device via the web UI (client.meshtastic.com) with 4 channels
2) Used the iOS app to chat in the channel 0, 1, 2 & 3 for a few days and build up a history
3) used client.meshtastic.com to change the order of the channels
4) opened the iOS app again

### Expected Behavior

I would expect that the channels would keep the history based on the channel names, not on the channel numbers.

### Current Behavior

The channels that i rotated around now show the history of the channel that previously held the channel slot.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-02-25:
You can change the channel name, channels are deleted and re-added each time you connect.  All mesh packets are saved with a channel index, if you you use a different client to alter that order I can't do much about it.

AssetBurned, 2024-02-25:
So the chat messages are stored on the Apple device? But not the corresponding name of the channel?

garthvh, 2024-02-29:
Yeah nothing matches on name, I updated the delete channel function to also remove all the messages for that channel.

closed 2024-02-29 by garthvh

AssetBurned, 2024-03-01:
maybe it would make sense to store messages on the Apple device with a reference on how the channel is called and the radio identifier ?

that would allow o slowly build up support for having two radios with a different set of chat history.

garthvh, 2024-03-02:
This is not supported by the firmware.

AssetBurned, 2024-03-23:
by firmware you talk about the software running e.g. on the Helltec, correct?