meshtastic/Meshtastic-Apple #529: 🐞 [Bug]: App freezes when accessing Route Recorder with no location permission

clwgh, 2024-02-27:
### Firmware Version

2.2.24

### What did you do?

Meshtastic iOS app `2.2.24 (828)`. iOS `17.3.1`. iPhone SE 3rd Gen.

Install the app and deny location permission

At any point after that, in the Meshtastic app, Settings > Route Recorder will freeze the application.

### Expected Behavior

Expected: the Meshtastic app should handle the lack of permission gracefully

### Current Behavior

Current behaviour: The Meshtastic app freezes and needs to be swiped away and restarted

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-03-10:
Works fine set to never, I can't replicate this.

closed 2024-03-10 by garthvh

clwgh, 2024-03-10:
To clarify for future visitors, this issue existed in Meshtastic app 2.2.24 (828) running on iOS 17.3.1 which were current versions when this issue was opened. Two weeks later, when this was closed, we are on versions 2.2.26 (1) running on iOS 17.4. In this combination a blank map appears and the app no longer locks up.