meshtastic/Meshtastic-Apple #531: 🚀 [Feature Request]: Add 1 decimal place to distances in Node view

clwgh, 2024-02-28:
### OS

iOS

### Description

In the current version of the iOS app (`2.2.24 (828)`), node distances are shown as integer numbers, eg **13 miles**. Would it be possible to increase the accurancy of this by one decimal place please, eg **13.4 miles**.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Thankyou for your work on this app.

garthvh, 2024-02-28:
Already does 
[image]

closed 2024-02-28 by garthvh

clwgh, 2024-02-28:
I was only seeing nodes > 10 miles away and none of those have a dp, so wasn't aware that single digit distances have a dp. Thanks.