meshtastic/Meshtastic-Apple #538: 🐞 [Bug]: Unable to Disable channel until Key is valid.

clwgh, 2024-03-02:
### Firmware Version

2.2.26 (1)

### What did you do?

- iPhone SE 3rd Gen
- iOS `17.3.1`
- Meshtastic app Version `2.2.26 (1)`

When creating a new channel, the _**Key**_ is checked against the _**Key Size**_ to ensure that it is a valid key. If it is not a valid key then the _**Key**_ field is bordered in red and the **Save** button is greyed out. This makes sense; it is ensuring that the channel has a valid key.

When going into _**Settings > Channels > an existing channel**_ with the intent to Disable (and thus delete) an existing channel, this same check takes place. Since the default _**Key Size**_ may not be correct for that channel's _**Key**_, the **Save** button is greyed out.

This means that a new _**Key**_ has to be generated for the the currently selected **Key Size** before the channel can then be Disabled and deleted.

### Expected Behavior

Expected behaviour:
- Go in to _**Settings > Channels > an existing channel**_
- Select Disabled
- Save

### Current Behavior

Current behaviour:
- Go in to _**Settings > Channels > an existing channel**_
- Select Disabled
- Unable to Save as button is greyed out
- Hit Refresh to generate a new key with whatever the currently chosen Key Size is
- This removes the red border and enables the Save button
- Save

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Only a minor bug but worth mentioning in case it's useful. Thanks for your work on the app.

garthvh, 2024-03-02:
So you want to disable the validation if you are disabling the channel and entered an invalid key from another client?

clwgh, 2024-03-02:
The flow here was that I set up some test channels today and experimented. Afterwards I wanted to delete the channels, which is done by going into each channel's config screen in turn, changing it from Secondary to Disabled, and Saving.

I was unable to do this because the channel config screen defaulted to a particular key size and I happened to have used a different key size for these channels. As a result the "invalid key" logic was triggered, preventing the Save button from working, despite the fact that I am deleting the channel.

The workaround is to use the "Generate new key" icon to populate the key with one which is valid for the default selected key size. This makes the "invalid key" logic happy and enables the Save button so the channel can then be deleted.

It totally get the logic and it makes sense when creating a new channel or editing an existing one to keep, but it doesn't make much sense when selecting Disabled since the channel is going to be deleted, so it becomes a blocker.

The fix would be that selecting Disabled bypasses the key logic and enables the Save button (perhaps changing it to read "Disable" instead of "Save")

closed 2024-03-07 by garthvh

clwgh, 2024-03-07:
Is the correct process to continue to generate a new key in order to delete a channel? Does not seem logical but I may be missing something.

garthvh, 2024-03-07:
They key is invalid, the validation is correct, the app where you made the invalid key needs a bug report

clwgh, 2024-03-07:
**The key was made in the Meshtastic iOS app.** It becomes invalid when the Meshtastic app later defaults to a different default Key Size when re-entering that channel's config. This means the channel cannot be deleted until a new key has been generated. That's the bug.

garthvh, 2024-03-07:
It loads the existing key size when you edit, I can't replicate this issue with generated keys

clwgh, 2024-03-07:
Here are the steps to reproduce in Version 2.2.26 (1) on iOS.

- Launch the app
- Settings > Channels > Add Channel
- Give channel name, select Key Size `1 byte` for a random `XX==` key
- Channel is `Secondary`, Save and the channel is created

Close the app and swipe the app away in iOS to simulate some usage and time passing.

Now to delete the channel. This will be done by editing it and setting it Disabled.

- Launch the app
- Settings > Channels > select the earlier channel name
- The default Key Size is either `256` or `128 bits`, **thus the previously selected 1 byte key is highlighted in red, because it is not a valid 256 or 128 bit key**.
- In order to delete this channel a new key has to first be generated so that it matches the currently selected key size.
- Once that is done, the red highlight disappears and the channel can now be set to `Disabled` and Saved, deleting it.

This may be an edge case where a) it's only happening with a 1 byte key, and b) the app has to be closed and reopened, and this triggers the errant behaviour.

Can you confirm if you are able to replicate this please?