meshtastic/Meshtastic-Apple #541: 🐞 [Bug]: GPS not sent to device

LanWolf, 2024-03-02:
### Firmware Version

2.2.23

### What did you do?

I disabled gps pass thru in the app setting a few miles before I got home last Thursday to prevent the gps expose my home location. 

After updating the app to 2.2.25 and later to 2.2.26 Friday,  today Saturday I remeabled gps pass thru  after drivig a few miles, but t keeps sending the last location from Thursday.

Restarting the device and phone did not help. 
Now it has no gps location, and does not transmit any location.

### Expected Behavior

I would expect the gps to pick up when enabling pass thru in the app

### Current Behavior

Does send no location or the last know location before updating the app.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Even went back to firmware form 2.2.24 to 2.2.23 to,check is it was few related.

nukemrules, 2024-03-02:
Same issue here. 
Even when Smart Position is on, the location isn’t being updated at all when moving or even when standing still on a new location. The position log is sporadic being updated, it mostly waits for the set interval. But when de app isn’t active at that time, the location is not updated. 
Also when you entirely close the app (from the active apps list) and reopen it, the location is not being updated (this was working before).

Adrelien, 2024-03-03:
Same here after updating IOS App, location stuck

MartijnVanTongeren, 2024-03-03:
Me too 2.2.23 beta, IoS app vrsion 2.2.26 (1)

aptonline, 2024-03-03:
I thought I was loosing my mind, same here. Location not updated on new node from iOS app no matter what settings I change. Running latest beta firmware (2.2.23) and iOS app up-to date.

fumpT, 2024-03-04:
Same on 2.2.22 here. Viewing the Bluetooth-connected node log shows new events but with stale coordinates. 

Curiously, Settings > Route Recorder *does* appear to update.

lostngone, 2024-03-04:
I am running a T-Lora T3 V1.6.1 running Firmware 2.2.21. iOS client version 2.2.26(1).
I am seeing something similar. What I have set is, under "App Settings" I have "Share location" and Smart position" both on. Under "Position Config" I also have "Smart Position" on. Under iOS settings for the Meshtastic app, "Location" I have it set to "Always". Also I keep the app running at all times.

eliot-trustm, 2024-03-05:
App version 2.2.26, Heltec v3 firmware 2.2.19  - also getting the same problem.  Location was working, but has now stopped.

Adrelien, 2024-03-05:
DIY 2.2.22 
Latest IOS 
Map Not Working 
Nodes do not show Node Map, Position Log
Pax Counter has no idea where it send
I can only send messages for now

Spud440, 2024-03-05:
Me too! 
Heltec V3
IOS app ver. 2.2.26
FW ver. Same results after trying 2.2.21, 2.2.22 & 2.2.23
Seems like it all started with one of the recent IOS Meshtastic app updates.

Adrelien, 2024-03-05:
> > I appreciate Meshtastic is open source but would be helpful to get some developer engagement on GitHub even if it’s just to acknowledge this is going worked on.
> 
> There's primarily a single developer that handles the Apple applications. It's important to remember that all of our developers volunteer their time. They're not going to be able to instantly respond to every issue/request. As soon as they're able to fix this, they will. In the meantime I could suggest downgrading to an earlier version until it is fixed.

We would love to downgrade but apple won’t let us. If you can roll back the version from app store it’s better

LanWolf, 2024-03-05:
> There's primarily a single developer that handles the Apple applications. It's important to remember that all of our developers volunteer their time. They're not going to be able to instantly [...]

sadly, on iOS you cannot downgrade, so we are stuck with this big issue until it is fixed..

altough I really respect the voluntary part, I would also expect a certain commitment.  
And a big bug like this without any response in days ? That leaves me with some questions about this project’s development continuity in the future.
If the current dev stops, meshtatsic just stops for iOS users ? 
there are many people buying (expensive) hardware use this software and then they will be left in the cold with non working devices ?
That makes me feel worried.

tropho23, 2024-03-05:
> sadly, on iOS you cannot downgrade, so we are stuck with this big issue until it is fixed..
>
> altough I really respect the voluntary part, I would also expect a certain commitment. And a big bug like [...]

This is a free project that you have chosen to use, and as stated before maintained by the generous free time and effort by *volunteers*.  You have also volunteered yourself to use it, free of charge, and free of any obligation to help maintain or enhance it.  There is no _commitment_ that anyone owes anyone else here.

If this is unacceptable to you feel free to choose another mobile device (Android?), or another open source mesh project.

Sm1thson, 2024-03-06:
> altough I really respect the voluntary part, I would also expect a certain commitment. And a big bug like this without any response in days ? That leaves me with some questions about this project’s [...]

All the code is online so unlike when google etc pull the plug on projects, anyone can keep this project going.

RCGV1, 2024-03-07:
I will try to find the cause of the issue. Yesterday I made it so I was able to push the code to my iPhone so I can test any changes I make. Probably has something to do with the newer changes so will look through those.

Adrelien, 2024-03-07:
> I will try to find the cause of the issue. Yesterday I made it so I was able to push the code to my iPhone so I can test any changes I make. Probably has something to do with the newer changes so will look through those.

First of all regardless of what the others said, thank you so much for putting your experience into this app. Second yes I think it’s regarding the last changes and updates. Last connect I had with the node was 27-03-2024 that the last date location was working. Lastly, just calm angry people down is there a way to rollback to older versions? 

I wish we could it ourselves but Apple doesn’t allow to install older version.

ThomasF98, 2024-03-07:
Thank you for checking into the iPhone Meshtastic App GPS problem Benjamin!

Adrelien, 2024-03-07:
> > I will try to find the cause of the issue. Yesterday I made it so I was able to push the code to my iPhone so I can test any changes I make. Probably has something to do with the newer changes so will look through those.
> 
> Pretty sure the issue is actually "resolved" in the TestFlight build. It's just the Apple Store that seems to have the issue. I imagine we can expect an update there as soon as Garth is able to.

Where I can find the test flight link?

RCGV1, 2024-03-07:
> First of all regardless of what the others said, thank you so much for putting your experience into this app. Second yes I think it’s regarding the last changes and updates. Last connect I had with [...]

I have only made one contribution to the iOS app before so I am not the one you should be thanking.

geeksville, 2024-03-11:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/position-greyed-out/10899/4

aptonline, 2024-03-11:
I can only comment on my own experience and situation but the latest TestFlight version of the app solved the GPS issue for me.  It’s worth pointing out that there has been no change to the GPS code so your mileage may vary.

aptonline, 2024-03-12:
The TestFlight version has now been released in the AppStore.  Are people seeing the GPS issue fixed for them?

natehudson, 2024-03-12:
Seems fixed for me. I haven't been able to forward my phone's GPS to a new Heltec V3. 

I upgrade the iOS app to `2.3.0 (842)` from the AppStore. Turned on "Smart Position" again and now see a proper fix and `Position Log` from the Heltec V3 from my iPhone's GPS.

Thanks!

ThomasF98, 2024-03-12:
I upgraded to the lastest Meshtastic App in the Apple store this morning. I did not update the firmware on my WisBlock. From my very limited testing, the new Mestastic App does once again appear to be properly allowing my WisBlock without built-in GPS to retrieve the GPS location coordinates from by iPhone. Big thank you to Garth Vanderhouwen and the rest of the Meshtastic team for getting this issue straightened out!

JayJayJohnson, 2024-03-12:
> The TestFlight version has now been released in the AppStore. Are people seeing the GPS issue fixed for them?

Works like a charm. THX

closed 2024-03-12 by thebentern