meshtastic/Meshtastic-Apple #548: 🐞 [Bug]: iOS App - Primary Channel Key Randomly Generated On First Edit

Xaositek, 2024-03-08:
### Firmware Version

2.2.24

### What did you do?

- Configure Channel 0 with an alternate name and encryption key
- Configure Channel 1 as default with AQ== key
- Save and ensure radio reboots and is functional
- Close (and force close) the iOS app
- Open iOS app
- Go to Settings -> Channels
- Open the Channel 0 entry and observe encryption key is different, Save Button is immediately active
- Go back to Channel List screen
- Open the Channel 0 entry and observe encryption is is now correct, Save Button is not active

### Expected Behavior

Channel 0 encryption key should always be correct, unless changed

### Current Behavior

Channel 0 encryption key is incorrect upon initial review, subsequent reviews do show the correct key

### Participation

- [ ] I am willing to submit a pull request for this issue.

hatesyourface, 2024-03-08:
I should have posted this the other day, a few others recreated this issue the other day when I mentioned it, but not everyone could. It's been this way for a while. 

Any channel that isnt using the default key initially shows an incorrect key, then subsequent views correct them.

closed 2024-03-12 by garthvh

clwgh, 2024-03-16:
I've seen this just now with app 2.3.0 (842) and Heltec V3 firmware 2.2.24.

Primary channel is LongFast with the default key. Secondary channel is a private one with a 256 bit key which was created on an Android device and shared with me and added using the web client.

With the app not running at all, launching the app, waiting for it to pair then going to Settings > Channels > the secondary channel in question, the Key shown is not the one that was added. Swiping that screen down to make it go away, and then re-entering the same channnel, the Key shown is now correct, and is on any subsequent swipe/re-enter.

Once the app is closed and swiped away in iOS, this bug can be repeated. It's not generating a new key, merely presenting a new key visually and not rendering the existing key on the first time the secondary channel settings are entered.

Presumably if an change was made, unrelated to the Key, and saved, this bug would have the effect of silently replacing the good Key with the randomly shown one.

Possibly related is that when the channel settings are opened the first time (with the wrong Key shown), the screen rolls up from the bottom and stops near the top with the primary channel showing through the gap. On subsequent views (with the correct Key shown) the page rolls all the way to the top with no gap. Mentioned in case it's relevant to the display logic going on with the Key.