meshtastic/Meshtastic-Apple #550: 🚀 [Feature Request]:  Wifi Connection between iPhone and Device

danielgrimard, 2024-03-10:
### OS

iOS

### Description

Can we have the option to connect to the hardware via Wi-Fi? I saw that this option is available on Android.
This feature will be helpful if we have one device and want to connect it to an MQTT server. 

### Participation

- [X] I am willing to submit a pull request for this issue.

garthvh, 2024-03-10:
You can use the MQTT client proxy to connect to a MQTT server, this would require a huge refactor for limited benefit, ESP32 wifi is unreliable and power intensive.  Not on my roadmap.

closed 2024-03-10 by garthvh

danielgrimard, 2024-03-10:
Thanks for your answer.
And BTW, you guys are doing an excellent job for the Meshtastic Project.

AssetBurned, 2024-03-23:
That is kind of a dup of https://github.com/meshtastic/Meshtastic-Apple/issues/506

smiba, 2024-06-23:
Would greatly appreciate this, as it would mean I can use the Meshtastic app on MacOS and have it send me notifications. Bluetooth unfortunately does not have enough range, and does not support multi-client.