meshtastic/Meshtastic-Apple #553: 🐞 [Bug]: GPS from app is sent to device despite being disabled

clwgh, 2024-03-12:
### Firmware Version

2.3.0

### What did you do?

Meshtastic app 2.3.0 (842)
iOS 17.4 iPhone SE 3rd gen
Heltec V3 running Alpha firmware 2.3.0

The Heltec is manually configured with a fixed location (CLI `--setlat` and `--setlon`).

The channel has Positions Enabled and Precise Location both Enabled to ensure that the fixed location is sent to the primary channel.

The app has GPS access and Settings > App Settings > Share location set to Disabled so that the phone's location is never sent to the Heltec, because that would override the fixed values configured previously and cause the phone's location to be sent to the channel.

Nevertheless this is exactly what happens. Once the position broadcast interval is reached, the phone's location is sent to the channel and appears on the map. Therefore the app has already sent it to the Heltec over bluetooth.

The phone's location values are confirmed as genuinely being in the Heltec by connecting the Heltec to the CLI and looking at `--export-config`.

The only way for them to have go there is for the app to have sent them. This all worked as expected in the previous version of the iOS app.

I suspect it may have something to do with whatever fix was put in place for https://github.com/meshtastic/Meshtastic-Apple/issues/541 which related to GPS not being sent when it should. Now it is being sent when it shouldn't.

### Expected Behavior

The app should not send the phone's GPS location to the Heltec if Settings > App Settings > Share location is Disabled. The channel's Positions Enabled setting is just to configure the Heltec to send the values of its Lat and Lon registers to the channel. It should have no bearing on whether the phone's location should be sent to the channel via the Heltec.

### Current Behavior

The app is overriding the Share location Disabled setting, possibly because the channel's Positions Enabled setting is being interpreted as meaning to send the phone's location to the Heltec, possibly as part of the fix for https://github.com/meshtastic/Meshtastic-Apple/issues/541.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-03-12:
Disabling the GPS on your hardware does not disable the position module.

closed 2024-03-12 by garthvh

clwgh, 2024-03-12:
Okay, you've skimmed this bug report and fired off a response and closed it. Your response has no relation to what is being reported here. The app is sending the phone's location to the Heltec, wiping out the manually configured location settings in there, despite the phone's setting for that being set to Disabled.

I spent quite a while gathering all that data for you, it is frustrating to have it just closed without understanding the report properly.

This is a reproducable bug and was not present in the previous version of the app.

Please re-open the report and work through it step by step.