meshtastic/Meshtastic-Apple #554: 🚀 [Feature Request]: Make Fixed Position respect Share Location

clwgh, 2024-03-13:
### OS

iOS

### Description

Meshtastic app 2.3.0 (842)

This is a feature request relating to app behaviour which causes a problem with manually configured locations.

## Summary

This is a followup from https://github.com/meshtastic/Meshtastic-Apple/issues/553. I have determined the cause of the behaviour seen. It is caused by Fixed Position ignoring the value of Share Location and always sending the phone's GPS location, a) if Fixed Position is Enabled or b) if any other option in the Position Config screen is saved while it is Enabled.

If the node is being used with a manually configured location (`--setlat`, `--setlon` and `--setalt`) then Fixed Position needs to be Enabled. However the act of enabling it, or even changing any other setting on that screen, causes the app to grab the phone's GPS and overwrite the manually entered settings, even when Share Location is Disabled.

## Workaround

Set all options on the Position Config screen to their desired values, including Fixed Position > Enabled, and Save the settings. Then disconnect the node and connect to the meshtastic CLI. Set the required values:

`meshtastic --setlat xx.xxxx --setlon yy.yyyy --setalt zzz`

This will restore the correct values which the app overwrote during the act of enabling Fixed Position in the app. Once done, make sure to NOT change any settings in the Positions Config screen or else the phone's GPS location will once again overwrite your manual values.

## Feature request

Make

  `Settings > Position Config > Fixed Position > Enabled`

respect the value of

  `Settings > App Settings > Share Location > Enabled/Disabled`

## Details

Regardless of the setting of Share Location, when Fixed Position is Enabled, the app sends the phone's GPS location to the node, which wipes out any previously manually entered values in the node. This requires the node to be manually reconfigured with the CLI to restore the overwritten values.

This also happens if Fixed Position is Enabled and an unrelated position flag is changed, for example toggling Altitude to Disabled and saving. The fact that Fixed Position is Enabled, during a save, once again grabs the phone's GPS location and sends it to the node, wiping out existing values.

The previous version of the app did not do this.

## Desired behaviour

Keep the functions of sharing location and fixed position logically separate.

If Share Location is Enabled then the phone's GPS location should be sent to the node to update its Lat, Lon and Alt registers.

If Share Location is Disabled then the phone's GPS location should not be sent to the node, and used only locally within the app (eg for computing distances to other nodes in the Node List). The node will acquire its Lat, Lon and Alt via some other means, eg a GPS module or manually entered values via the CLI.

If Fixed Position is Enabled then set `fixedPosition: true` in the node.

If Fixed Position is Disabled then unset `fixedPosition`.

If any Position Flags are changed and saved, set the relevant decimal value for `positionFlags` on the node and do not use this to send the phone's GPS location to the node if Fixed Position happens to be Enabled.

Thanks for your continued work on this app, I hope this report is useful.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-03-13:
None of this has changed, this is how fixed position works. When you enable fixed position it sends a location to the device, there is text telling you this where you enabled fixed position.

On the node fixed position sends out the most recent position it ihas.

closed 2024-03-13 by garthvh

clwgh, 2024-03-13:
I am sorry but you are mistaken on this. I'm trying to be polite and help by being as detailed as possible, but you're just dismissing things without looking properly.

Fixed Position [means](https://meshtastic.org/docs/configuration/radio/position/#fixed-position):

> If set, this node is at a fixed position. The device will generate GPS updates at the regular GPS update interval, but use whatever the last lat/lon/alt it saved for the node. The lat/lon/alt can be set by an internal GPS or with the help of the mobile device's GPS.

My node is at a fixed position and I have set that position manually. I need to enable Fixed Position. But if I use your app to enable Fixed Position, **you are taking it upon yourself to also send the phone's location to the node**.

You also do this if any Position Config option is changed just because Fixed Position is Enabled.

Please STOP doing that, it is not your place to assume how the location data gets to the node when Fixed Position is Enabled. And don't forget that Share Location is Disabled and you are completely ignoring that. Imagine someone's surprise when they disable location sharing and configure a manual location, only to find later on their phone's location going out over primary.

That text says "but use whatever the last lat/lon/alt it saved for the node". I have configured that and you keep overwriting it. Please respect the Share Location setting, and please stop treating Fixed Position as "send the phone's location".

garthvh, 2024-03-13:
It sends a location to the device, this is not new functionality. The length of the issue is not really relevant.

garthvh, 2024-03-13:
[image]

if you don't want to use phone location at all that is an option as well

clwgh, 2024-03-13:
I need to use the phone's location in the app (and only the app) because that's how the app knows how far way nodes are in the Node List.

I do need Fixed Position enabled, as that tells the node to use it's stored Lat, Lon and Alt, even without access to a GPS source. I have programmed the correct values for Lat, Lon and Alt via the CLI.

Using the app to set the Fixed Position flag unfortunately wipes out those settings because you've decided you want it to mean that the phone's location will be used. And that's also despite Share Location being firmly set to to Disabled.

Can't you just make it work like it did in the previous version? Fixed Position did not wipe out stored values in that version, it simply set the FP flag and left it up to the node to manage how it gets values into its Lat, Lon and Alt registers.

clwgh, 2024-03-13:
> It sends a location to the device, this is not new functionality.

Fixed Position shouldn't send a location to the device, it should simply set `fixedPosition: true` on the node. The node already has the correct fixed location programmed into it.

garthvh, 2024-03-13:
Not really interested in having incomplete functionality in the app so that you can also use the CLI to spoof positions.  If you wish to disable the phone location functionality in the app that option is available to you.

clwgh, 2024-03-13:
Why are you silently sharing location with the node when Share Location is set to Disabled? Don't you think that's a terrible idea?

As for spoofing positions, what's the difference between me setting a nearby fixed location via the CLI (if the app supported that I would use it for that), or using the app's new Approximate Location setting to do it?

garthvh, 2024-03-13:
There is no silently sharing location, there are other feature that use location.

clwgh, 2024-03-13:
When Share Location is set to Disabled I expect my location to not be shared. Please rethink you UI.

clwgh, 2024-03-13:
> The firmware and app has plenty of location privacy options, use them.

Thanks but I'll use the most appropriate supported method for my use case. For now the workaround I posted gets around this UX silliness.

GUVWAF, 2024-03-13:
> Out of interest, how would you propose I set a fixed unchanging location and interact with the node using your app and the tools available?

Set a position precision per channel and let the app use the current location from your phone **once** to set a fixed position when you flip the switch. Leave “Share location” disabled. That it sends a new position if you change other position settings is a bit unfortunate, but with imprecise position this shouldn’t be a problem. If your node is really in a fixed position, it won’t change as you need to be in BLE range of the device.

Alternatively, just set your lat/lon **and** fixed position (and other position settings) with the CLI. 
The iOS app is not designed to be used in combination with the CLI; it doesn’t know whether the current position the device has is the one you want or not and spoofed locations are dis-encouraged. When you use imprecise location, other people know the range it might be in.

clwgh, 2024-03-13:
Thanks GUVWAF. I'm sort of doing that anyway as a workaround. Set Fixed Position to on, set the flags as I want them, then use the CLI to enter the desired values and then never use the Position Config screen again.

> That it sends a new position if you change other position settings is a bit unfortunate

It shouldn't be sending position at all because Share Location is off.

> The iOS app is not designed to be used in combination with the CLI

I'm not sure that's true, the node doesn't care where the settings come from. I'm using the app for most of the setup and the CLI for a couple of things the app doesn't support. The app is unfortunately now ignoring the Share Location setting and making its own assumptions about why Fixed Position is enabled. At least making no new changes in Position Config avoids this problem. Other than this annoyance the app works well and I'm grateful for the developer's work on it.

GUVWAF, 2024-03-13:
> The app is unfortunately now ignoring the Share Location setting and making its own assumptions about why Fixed Position is enabled.

“Share Location” is for a separate functionality (when you’re moving around and don’t have fixed position enabled). In my opinion, the text at “Fixed Position" is pretty clear: “If enabled your current phone location will be sent to the device”. 

Probably just best to use imprecise location from now on instead of spoofing via the CLI if you plan on configuring the device with your phone.

clwgh, 2024-03-13:
It has the same text in the previous version of the app but that didn't actively send the phone's location to the node when enabling it, or when saving another option on that screen where it was already enabled, which was the correct behaviour for that flag (Fixed Position should have no input on how the node's Lat, Lon and Alt become populated).

I took that wording to imply that it is simply setting `fixedPosition: true`, and that if Share Location was enabled, those would be the values in the node, otherwise they would be untouched by the app. I can see this was a wrong assumption and that the text is literal - activating Fixed Position in the app means the node is a phone location tracker (which ironically is not what fixed position is for at all).

I found this out yesterday when I discovered that instead of the very nearby position I had configured, the simple act of toggling Altitude on during the day had updated the node's location registers and it was now sending my precise location to LongFast and had been doing for some hours.

> Maybe @garthvh can let it send a position to the device only when it toggles Fixed Location from off to on?

I think that is the case already, hence the workaround of configuring most settings including position options in the app, correcting the location in the CLI and then never touching that screen again.

Other than this behaviour the app works well and is great for interacting with the node and other users.

garthvh, 2024-03-13:
> It has the same text in the previous version of the app but that didn't actively send the phone's location to the node when enabling it, or when saving another option on that screen where it was [...]

It did, this code is unchanged for over a year.

GUVWAF, 2024-03-13:
> activating Fixed Position in the app means the node is a phone location tracker 

It is not, only when you change position settings it will update it as you mention. But this is a non-issue anyway if you're using imprecise location.

clwgh, 2024-03-13:
> It did, this code is unchanged for over a year.

I suspect that I happened to first use the app while bug 541 was present and, once that was fixed, the observed behaviour returned, the first time me seeing it. The previous couple of versions of the app definitely did not do this.

> this is a non-issue anyway if you're using imprecise location.

Do all users (iOS, Android, web client, API, etc) see imprecise location and a circle of possibility, or is that a homebrew something done at the application level which is only intepreted within the iOS app, and any apps supporting the same protocol?

GUVWAF, 2024-03-13:
The firmware cuts off some of the bits (depending on the setting) from the position before it sends it out in combination with the precision setting. How other clients visualize this is up to them.

clwgh, 2024-03-13:
> How other clients visualize this is up to them.

Exactly. That's why I'm using a very nearby fixed precise location instead, as I prefer to place myself exactly on the map and not gamble that dropping dp's and precision does the job correctly. There's nothing at all wrong with that and I can see now the app does not support this feature of Meshtastic (only with forced assumptions), hence the workaround (for anyone finding this later on in the same situation).

garthvh, 2024-03-14:
The share location toggle in the app has no control over the position stored on the node.  The position precision is sent over the mesh and is the proper ways to send imprecise locations, no need for silly workarounds.

clwgh, 2024-03-14:
It's not a workaround, it's a valid configuration for a fixed node which your app doesn't support. No big deal, next time just say that so we can move on with a solution that does support it.

clwgh, 2024-03-14:
Location approximation is no more or less "spoofing" than setting a specific location. It's either right or it isn't. The best your app's approach can manage is 320m which is way too far away (over 1µs), and it supports placing me nearly 12km away! And, as GUVWAF points out, it requires the receiver to parse the location imprecision as intended.

In my case, when I'm done testing, I want to be able to set a precise location a few tens of metres away which is why I am setting a fixed location. Close enough to be good enough for my requirements but far away enough to be private. I can use the app for some of this and the workarounds for the rest.

We'll have to agree to disagree on this one, Garth. Thanks for your work on the app.

AssetBurned, 2024-03-23:
https://github.com/meshtastic/Meshtastic-Apple/issues/424

AssetBurned, 2024-03-23:
https://github.com/meshtastic/Meshtastic-Apple/issues/507

AssetBurned, 2024-03-23:
It isn't just about spoofing a position. Think of the situation that you want to place a node at a position where you (and your phone) is not allowed to be, but the node is. As in not everyone is allowed to access roof tops, water towers etc. In that case there should be the ability to set the precise location in the app manually and not rely on the phones location.

I mean we can also look at it from a very different perspective and simply say that the iOS app isn't meant for advanced setups and more be seen as an chat client with some additional functionalities that may or may not work as the user expects. Which I would find rather sad for various reasons, especially as i think the app as such is a great product you maintain Garth.

garthvh, 2024-03-23:
I don't think that is a real use case, there is no where you can take the node and not have your phone. People use manual entry to add Santa and vandalize maps.