meshtastic/Meshtastic-Apple #557: 🐞 [Bug]: App does not accept a key created by an Android device

clwgh, 2024-03-16:
### Firmware Version

2.3.0 (842)

### What did you do?

iOS app 2.3.0 (842). Heltec V3 fw 2.2.24.

I have the Primary channel set to LongFast with the Default key. I am trying to add a Secondary channel created by someone using the Android Meshtastic app. The channel has a name of the form ChannelName and a 256-bit key.

I go to Settings > Add Channel and enter the ChannelName. I select Key Size 256-bit and the app generates a random 256-bit key. I delete that key and paste in the correct key. However the field border is red and the Save button disabled because the app thinks the key is not valid. I have tried:

- copying and pasting the key
- copying and pasting the key in landscape so there is no wrapping the box
- copying the key and ensuring no possible control codes are present, eg end of line
- manually typing in the key very slowly and carefully
- manually typing in landscape mode so no wrapping

The key is just not accepted as valid. To make it work I had to use the node client app and enter it there. Having now been entered there and reconnected to the app, the app shows the channel and the key. I can even now edit the key and cause the border to turn red, and then re-type the correct characters and the red border goes away. The app is now happy with the key. It is just entering it in the app as the initial entry which doesn't work.

Someone in a forum has said it's a known bug and possibly something to do with hyphens being interpreted as underscores, or problems with pasting keys, but I have tested those and no joy.

It may be just certain keys, I'm certainly not saying that all Android keys behave this way on the iOS app. I do not have an Android phone or I would test to help further.

### Expected Behavior

Affected Keys are accepted as valid by the on initial entry during creation of the channel.

### Current Behavior

Affected keys are only treated as valid by the app after first being entered by some other means with the channel now already set up.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-03-16:
Some of the characters in the key are not valid.

clwgh, 2024-03-16:
I've asked the creator if they mind me posting it here for analysis, might be a day before they reply. It'd be good to know where it's wrong. It may be that the Android app is creating some invalid keys and the iOS app is correctly detecting and preventing them. It was happy to let me edit the key and invalidate it then edit it back again and accepted that no problem, once it was added via the web interface. I'm not sure if that does any kind of error detection.

garthvh, 2024-03-17:
https://base64.guru/tools/validator

clwgh, 2024-03-17:
The validator flags that the key is invalid:

`The following characters are not suitable for the specified standard: -`

The repair tool replaces the two `-` characters with `+` characters.

I'll point this out the the originator because this implies the Android app is sometimes creating invalid keys, or keys to a different standard (possibly base64url?). It may be a bug whereby the key wraps and a hyphen is inserted but this becomes copied as part of the key (although then I would expect the length to be incorrect too).

Either way it appears to not be an iOS app bug and the iOS app is correctly detecting and preventing the invalid key. Thanks for the info and the link to confirm.

Closing as not a bug.

closed 2024-03-17 by clwgh

clwgh, 2024-03-17:
> It's probably expected by the user that they can copy a key from one client to another and it should work in all of them.

A channel was set up using the Android app and the key shared, but the key appears to be a different form of base64 (perhaps base64url vs base64). Is there a firm standard for creating Meshtastic channel keys?

You're right, without some standard approach it is entirely possibly to inadvertently cut off part of the user base when trying to created a shared space at scale due to different client approaches. It could also just be an application bug in the key creation or presentation despite all apps following some standard.

garthvh, 2024-03-17:
The keys are base64, they are converted to base64url and back when used in a QR code.