meshtastic/Meshtastic-Apple #560: 🐞 [Bug]: Positions appear to not obfuscate if sent to alternate topic suggestions in MQTT

quade-ksp, 2024-03-19:
### Firmware Version

2.3.1 (860)

### What did you do?

Sent position using various alternate topics and was shocked when they revealed precise location on MeshMap.net.

Last seen	By node	On MQTT topic
3/18/2024, 9:59:56 PM	!xxxxxxxx	msh/2/e/LongFast
**3/18/2024, 4:37:34 PM	!xxxxxxxx	msh/US/CA/2/e/LongFast
3/18/2024, 2:00:52 PM	!xxxxxxxx	msh/US/2/e/LongFast**
3/17/2024, 9:21:45 PM	!xxxxxxxx	msh/US/2/c/LongFast

Was able to obfuscate properly by sending to 3/18/2024, 9:59:56 PM	!xxxxxxxx	msh/2/e/LongFast

### Expected Behavior

Obviously, with position set to something lower precision it should not matter which topic it goes to.  It should not in any way leave the device at the highest setting for precision if set lower, but something is obviously amiss.

I preformed the experiments a couple of times over the course of the day and for me it was repeatable.

### Current Behavior

Again, the precise location should never leave in any case if set to lower precision.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Would be happy to answer any further questions.

garthvh, 2024-03-19:
The channel determines the precision, not the mqtt topic

quade-ksp, 2024-03-19:
Absolutely understood that is how it’s supposed to work.

What I’m saying is there is a bug, that allows the precise location to be sent to MQTT if the topic is changed.

All you need do is test this yourself.  Obviously it will take several minutes to see the changes reflected on MeshMap, but they absolutely happened to me and rather than showing my approximate location, it showed my exact location.

Please, I urge you to look into it.

Paul

garthvh, 2024-03-19:
I just tested it and it works as expected, mesh map does not display the imprecise location circle

garthvh, 2024-03-19:
There are different messages, saving mqtt settings does not do anything to the channels

quade-ksp, 2024-03-19:
I’m very confused by this then.  I’m not saying MeshMap.net shows or doesn’t show the imprecise circle.  I’m saying it pinpoints to the exact house I live in when set to the alternate topics.

I’ll screen record my doing this and send a link to that to you tomorrow.  I know the issue is reproducible, but it does take more time than I have tonight to do that.

Thank you for your time so far on this.

Paul

garthvh, 2024-03-20:
I am not seeing any precise positions unless they are rnabled

jp-bennett, 2024-03-20:
> I’ll screen record my doing this and send a link to that to you tomorrow. I know the issue is reproducible, but it does take more time than I have tonight to do that.

Please do. We're having trouble reproducing consistently, and would very much like to nail this one down.

quade-ksp, 2024-03-20:
Yeah, I’m having issues reproducing as well.  I know what I saw at the time and was shocked by it.I’m beginning to wonder if this is really an issue with the MQTT database and how it’s being used by the MeshMap.net folks.I also wonder if it leaked out via a secondary channel which got sent to MQTT.Paul

garthvh, 2024-03-21:
I am going to close this as I tested the app positions pretty thoroughly and was never able to get a position to come through without doing it on purpose, I did find this https://github.com/meshtastic/firmware/issues/3450 which made me think it happened, but then when I zoomed in the pin was at the apartments across the street.

closed 2024-03-21 by garthvh