meshtastic/Meshtastic-Apple #562: 🐞 [Bug]: "Disconnect from MQTT" switch

AssetBurned, 2024-03-23:
### Firmware Version

App 2.3.1 (860) iOS

### What did you do?

sliding the off/on slider

### Expected Behavior

Staying aligned with the rest of the options on the page and not seeing that the description of the slider suddenly changes

### Current Behavior

When using the slider the description now changes to "Connect to MQTT" using the slider again it changes back to "Disconnect from MQTT". This gives the impression that when "Disconnect from MQTT" is visible we are disconnected and when "Connect to MQTT" is visible we are also disconnected.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

This is really not an intuitive way. No other slider there changes the description. "Encryption Enabled" for example stays that way no matter if the slider is in off or on position.

I do understand the idea here might have been to make it easier for people to understand how to disconnect from the MQTT, but those sliders are a standard functionality in iOS which should have an consistent behaviour across all apps. Maybe a fine print text would make more sense similar to the "Utilitizes the network connection on your phone to connect to MQTT" there could be an "You are now connected to the MQTT service" or "You are NOT connected to the MQTT service.".

garthvh, 2024-03-23:
It does what it says, lets you connect and disconnect from mqtt

closed 2024-03-23 by garthvh

AssetBurned, 2024-03-23:
so you are planing to change the other switches the same way?

garthvh, 2024-03-23:
They don't do the same thing, I am going to leave it as is. It controls the actual MQTt connection and does not save to the device.