meshtastic/Meshtastic-Apple #578: 🚀 [Feature Request]: Ability to connect other Nodes via DeepLinking or Scanning QR Code.

AakifNadeem, 2024-04-02:
### OS

iOS

### Description

For now, we are unable to scan QR from other iPhones connected to another node (Just like there is a scan QR option in Android). Also, the app is not connected via deeplink when we are opening the shared QR Code link.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-04-02:
You scan the QR code from the camera

AakifNadeem, 2024-04-03:
@garthvh when scanning from camera I get this screen on browser.

[image]

garthvh, 2024-04-09:
You may need to reboot your phone, the url is registered with apple and opens the app automatically

closed 2024-04-09 by garthvh