meshtastic/Meshtastic-Apple #589: [Feature Request]: Allow swipe left to cancel BLE connection during a doomed to fail retries connection flow

thebentern, 2024-04-08:
### OS

iOS

### Description

During a normal connection flow for BLE, I'm able to disconnect during the connection or after the successful connection with the normal swipe left behavior but when connecting to a device with bad auth (typically after a factory reset when the phone uses an expired auth token to attempt connection), I'm unable to escape the N out of 10 retries flow. 

I'm forced to wait for the flow to complete or close the app (which is what I currently do) and go forget the device. The other issue this causes is that I can't cancel the bad connection attempt flow and connect to _another_ device in BLE range. If I close the app and reopen, it just retries the bad connection flow again if that device was the last one I successfully connected to. I've also had this be an issue on the very edge of BLE range where the phone is never quite able to seal the deal on the connection flow.

This issue is admittedly somewhat niche and probably only relevant to developers and power users doing a lot of crazy things to devices, but I wanted to capture it in case someone wants to take a look. Seems like a good starter issue

### Participation

- [ ] I am willing to submit a pull request for this issue.

RCGV1, 2024-04-12:
I created a preliminary PR for this, still changing it a bit but I also was able to fix the infinite connecting issue.

https://github.com/meshtastic/Meshtastic-Apple/pull/594

garthvh, 2024-04-16:
I added a method to get out of the loop, which may fix a BLE error crash as well, in test flight

closed 2024-04-16 by garthvh