meshtastic/Meshtastic-Apple #590: 🚀 [Feature Request]: Add possibility to set the Timezone info on the node

caveman99, 2024-04-08:
### OS

iOS, iPadOS, macOS

### Description

Recently the time display on the firmware screen (and logfile) for the node was made aware of timezones. Please add the ability to set via config.device.tzdef. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

The required format is the raw timezone description. Added bonus would be for the user to be able to select a more legible format  from a dropdown. A translation table can be found on this link: https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

garthvh, 2024-04-08:
That seems bad, I need UTC in the app.

garthvh, 2024-04-08:
I don't want to manage it at all, apple is way better at this than me or the posix solution.

This is in TestFlight under device config, it builds the value from your timezone.

caveman99, 2024-04-08:
There may be more points where this needs to be taken into account, not many but I think I can find them. From the top of my head telemetry and Position values
 The current fix should correct Position timestamp.

garthvh, 2024-04-08:
Yeah looks good here

garthvh, 2024-04-09:
In app review, fixed by https://github.com/meshtastic/Meshtastic-Apple/pull/584

closed 2024-04-09 by garthvh