meshtastic/Meshtastic-Apple #591: 🚀 [Feature Request]: Notification when a new node is found

jclagreca, 2024-04-09:
### OS

iOS, iPadOS, macOS

### Description

Would be great to get a push notification on the app when a new node is detected.  This would be very useful for war driving scenarios, or just to know when a new user was around.  

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-04-11:
> Also, we'll want to make this behavior an opt-in via an App setting, as there are large MQTT based meshes that would blow up user's push notifications.

I added a settings bundle so we can put a bunch of notification settings in the root app settings panel.

garthvh, 2024-04-11:
Closed by https://github.com/meshtastic/Meshtastic-Apple/pull/593

Waiting for test flight approval

closed 2024-04-11 by garthvh