meshtastic/Meshtastic-Apple #596: 🚀 [Feature Request]: A different colour for each message status text

clwgh, 2024-04-10:
### OS

iOS

### Description

### Minor feature request
iOS app 2.3.4 (900)

The message status texts have different colours:

Max Retransmission Reached - red
Acknowledged - grey
Waiting to be acknowledged. - orange
Acknowleged by another node - orange

Could these last two have different colours please, as the small text, same orange colour and identical length makes them difficult to determine without getting up close to study the wording of the text (the others can be determined at a glance from a distance by the unique colour without needing to read the text).

Thankyou.

### Participation

- [ ] I am willing to submit a pull request for this issue.

AssetBurned, 2024-04-11:
Thought about accessibility… would it make sense to have custom colors to allow adjustment for color blind or is that somehow else covered?

clwgh, 2024-04-11:
> Thought about accessibility… would it make sense to have custom colors to allow adjustment for color blind or is that somehow else covered?

It would be simpler to select colour-blind-friendly colours as standard, as that's just a colour change in existing code rather than adding new switches which would be a separate feature request.

This request is just as a minor request to change one of the orange colours to help distinguish between those two orange messages.

garthvh, 2024-04-16:
purple implicit add added to TestFlight 2.3.5, awaiting app review

closed 2024-04-16 by garthvh

clwgh, 2024-04-16:
That's great, many thanks.