meshtastic/Meshtastic-Apple #601: 🚀 [Feature Request]: Standalone watchOS App

photogeen, 2024-04-13:
### OS

iOS

### Description

Hi, how complicated would it be to create a stripped down version of the Meshtastic App for the Apple Watch, or maybe a companion app. The simple things that would be needed is:

- BT connection from Watch to node
- Text messaging 
- sharing GPS location from Watch to node

There are a lot of non cellular Apple Watches out there. And with this it would be possible to stay connected in a group without the need of an iPhone. Just a couple of cheap SE Watches and nodes.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-04-16:
Last time I looked core bluetooth can't run in the background on the watch.

closed 2024-04-16 by garthvh

vvuk, 2025-04-06:
I think this might be (now?) possible -- this Apple example seems to contain all the necessary bits: https://developer.apple.com/documentation/watchkit/interacting-with-bluetooth-peripherals-during-background-app-refresh

If I'm understanding the Meshtastic device API + Apple's watchOS support:
- the watchOS app should be able to function normally as long as the app is open; it has the ability to connect to bluetooth LE devices, which should be all that's needed. (Ideally at some point the phone app and the watchOS app should communicate, so the complex configuration can be done on the phone, and the watch just selects from predefined configs)
- `CBCentralManager` an be used to discover the meshtastic device, and `CBPeripheral:setNotifyValue:for:` can be used to indicate that the app wants notifications whenever the meshtastic `fromnum` characteristic's value changes
- the app will get `CBPeripheral:didUpdateValueFor:` notifications, and in the handler it can do the FromRadio reading from the device and display notifications etc.

does that make sense, am I missing something? I may try to implement this as this would be extremely handy!

garthvh, 2025-04-06:
Would need to be a workout app and disconnects if the app goes into the background last I looked so not real practical.

vvuk, 2025-04-06:
I think that's only if you actually want to use "Extended Runtime", where you need your app to stay alive longer than normal. I don't think this is that, this lets the OS know what BLE device/characteristic you want to listen to, and then your app goes back to sleep until it gets a notification. I'll do some testing, should be possible to hack something up pretty quickly.

garthvh, 2025-04-06:
Meshtastic really does not sleep, what would the goal be? As is the watchOS BLE is not a good fit, maybe there will be some enhancements at wwdc this year.

DanDawson, 2025-06-30:
I just wanted to add that I would also be interested in this and wanted to subscribe to the issue in case any progress gets made. Side note: There are several comments above that Bluetooth disconnects if not in the foreground, but I don't believe that is true: My heart rate monitor (3rd party) and AirPods work just fine with my Apple Watch even when their app is not in the foreground.