meshtastic/Meshtastic-Apple #602: 🐞 [Bug]: Text input box hidden by on screen key pad - iPad - iOS17.4.1

senditron2000, 2024-04-14:
### Firmware Version

2.3.3

### What did you do?

When entering text to send, the iOS key pad comes up and covers the text input box and send button. The input text box does not move up above the on screen keyboard. You can not see what you are typing and also do not have access to the send button until you press the keyboard down button to hide the keyboard.

### Expected Behavior

Text input box and send button should move up as the on screen keyboard comes up.

### Current Behavior

Text input box and button hidden by on screen keyboard.

### Participation

- [ ] I am willing to submit a pull request for this issue.

senditron2000, 2024-04-17:
Ipad 7th Gen. Text size is standard (middle of the slider). Tried other text sizes, didn't fix it.

https://imgur.com/vz6kpRp

garthvh, 2024-06-07:
Try upgrading your OS and app, I am not seeing this on a 10th get iPad or a mini.

closed 2024-06-07 by garthvh