meshtastic/Meshtastic-Apple #605: (iPhone) Device metrics shifted to left, making batt% unreadae

MansM, 2024-04-17:
### Firmware Version

2.3.5

### What did you do?

go to nodes, select node, device metrics log

[image]

### Expected Behavior

Readable battery percentage

### Participation

- [ ] I am willing to submit a pull request for this issue.

purge98, 2024-04-19:
Uptime is always 0s.

thebentern, 2024-04-19:
> Uptime is always 0s.

You need to update to the latest firmware alpha to report uptime. It was just added last week.

paoloponzano, 2024-04-19:
This happens for me too

purge98, 2024-04-20:
The node map restriction distance is too limited. I live over 5500 kms from NZ. Please allow up to 16,000 kms so I can see the whole world. Why the limitation?

MansM, 2024-04-20:
Can you guys make separate issues for your issues, helps maintainers to be able to close fixed issues. 

And me getting notifications for the additional comments.

garthvh, 2024-04-22:
Released in 2.3.6

closed 2024-04-22 by garthvh