meshtastic/Meshtastic-Apple #616: 🚀 [Feature Request]: Notification settings per channel/type

tekstrand, 2024-04-23:
### OS

iOS, iPadOS, macOS

### Description

It would be neat to be able to subscribe to notifications for one channel but not another, or DMs but no channels. What made me think of this is an extremely busy Default/Long Fast channel where tons of people are just saying "copy" while I am having a separate conversation that I do actually want notifications for.

### Participation

- [ ] I am willing to submit a pull request for this issue.

tekstrand, 2024-04-23:
@garthvh let me know why this one is a bad idea ;)

garthvh, 2024-04-23:
You can mute DMs, doing it for channels is harder as they get deleted and re-added on every connect, so I had to remove it, I would like it back.

garthvh, 2024-04-23:
@tekstrand I think I could do it if we added mute here https://github.com/meshtastic/protobufs/blob/eade2c6befb65a9c46c5d28ae1e8e24c37a1a3d0/meshtastic/channel.proto#L94

tekstrand, 2024-04-23:
Oh good point RE: location precision on the module. Right up the same alley. If you'd like me to help, let me know how!

garthvh, 2024-04-27:
We need a protobufs issue

tekstrand, 2024-05-01:
So quick! I stepped away for a few days and it's already in alpha!

closed 2024-05-05 by garthvh