meshtastic/Meshtastic-Apple #617: 🐞 [Bug]: Meshtastic Leaks Precise Location from Imprecise Secondary Channel

Coopersmith-24601, 2024-04-23:
### Firmware Version

2.3.6.7a3570a

### What did you do?

Please see attached .pdf

[Meshtastic Precision Location Bug.pdf](https://github.com/meshtastic/Meshtastic-Apple/files/15081893/Meshtastic.Precision.Location.Bug.pdf)

### Expected Behavior

Sharing a location from any channel SHOULD NOT in any way ever send the location in anything other than that channels defined precision.

### Current Behavior

Sharing location from a Secondary channel with a less than precise location sends the precise location to the other user.  Although the second user's device with draw an imprecise location circle, the center of that circle and the Node ID sent will appear in the precise location of the first user.

### Participation

- [ ] I am willing to submit a pull request for this issue.

daviesgeek, 2024-04-23:
I seem to be able to reproduce this as well. Primary channel seems to work fine to set imprecise location, but the secondary channel does not; it sends the exact location but marks it as imprecise, so while it shows up with a circle on the app and the web maps like meshmap or Liam Cottle's map, the center of the circle is my exact location

garthvh, 2024-04-24:
All the send position functions look up the channel and use the precision settings.

garthvh, 2024-04-24:
@jp-bennett the positions appear to have the position precision value set but the coordinates are not getting reduced, the same message sending function is working when the positions are sent with the timer

jp-bennett, 2024-04-24:
I think getPositionFromPhoneGPS() is building the whole packet, and the device firmware is just forwarding it through as it's written. So the two code branches in getPositionFromPhoneGPS() would need to also do the position dilution steps.

garthvh, 2024-05-02:
The firmware has been updated, pins send in DMs will still be sent out with full precision, but broadcast positions will use the channels position precision from the firmware.

closed 2024-05-02 by garthvh