meshtastic/Meshtastic-Apple #629: 🐞 [Bug]: Settings > User > Call Sign issues

gitToad, 2024-05-02:
### Firmware Version

2.3.6 (917)

### What did you do?

Mac App Settings > User > Call Sign

### Expected Behavior

1. now only allows short names (used to be more, shame to lose the extra letters for descriptive names and need to update node names / docs for people now...)

2. says it can be up to 8 bytes long. Actually only allows 6

3. perhaps "8 characters long" is more user friendly than "8 bytes long" (detail) ?

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-05-02:
8 bytes is not 8 characters, long name is used for call sign short name is only 5 bytes

garthvh, 2024-05-05:
It was restricted to 6 bytes, which is less than six characters, it has been updated to allow 8 bytes

closed 2024-05-05 by garthvh