meshtastic/Meshtastic-Apple #635: 🐞 [Bug]: MQTT does not connect with TLS enabled

Flothoger, 2024-05-07:
### Firmware Version

2.3.7

### What did you do?

I am running a selfhosted Mosquitto server with LetsEncrypt TLS certificates. Connecting a generic MQTT Client, like MQTTX, works fine with not TLS / SSL errors.

When connecting using the iOS or macOS app the connection times out. In the app I get the following error: `kCFStreamErrorDomain SSL-Error -9806.`
When looking at the Mosquitto logs I see:
```
1715074957: New connection from 10.0.1.82:51076 on port 8883.
1715075025: Client <unknown> disconnected: Protocol error.
1715075408: New connection from <IP-removed>:56488 on port 8883.
1715075442: Client <unknown> has exceeded timeout, disconnecting.
```
I tried with TLS v1.3, v1.2 and v1.1. Changing it on the server side makes no difference.
I am not sure wether this might be a bug or user error, but since the connection is working with any MQTT client other than Meshtastic I think it might be either something with the app itself or how iOS handles TLS traffic.

### Expected Behavior

The MQTTS connection should succeed with no errors. On the Mosquitto server I would expect log entries like this:
```
1715075231: New connection from 10.0.1.82:51092 on port 8883.
1715075231: New client connected from 10.0.1.82:51092 as mqttx_24b4f0dd (p5, c1, k60, u'test').
```

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Node config:
```
# start of Meshtastic configure yaml
channel_url: <removed>
config:
  bluetooth:
    enabled: true
    fixedPin: 123456
  device:
    nodeInfoBroadcastSecs: 10800
    serialEnabled: true
  display:
    gpsFormat: UTM
    screenOnSecs: 900
  lora:
    hopLimit: 3
    region: EU_868
    sx126xRxBoostedGain: true
    txEnabled: true
    txPower: 27
    usePreset: true
  network:
    ntpServer: 0.pool.ntp.org
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    gpsUpdateInterval: 120
    positionBroadcastSecs: 900
    positionBroadcastSmartEnabled: true
    positionFlags: 811
  power:
    lsSecs: 300
    minWakeSecs: 10
    sdsSecs: 4294967295
    waitBluetoothSecs: 60
location:
  alt: 103
  lat: <removed>
  lon: <removed>
module_config:
  ambientLighting:
    blue: 124
    current: 10
    green: 122
    red: 109
  detectionSensor:
    detectionTriggeredHigh: true
    minimumBroadcastSecs: 45
  mqtt:
    address: mqtt.<removed>.de
    enabled: true
    encryptionEnabled: true
    mapReportSettings:
      positionPrecision: 12
    password: test1234
    proxyToClientEnabled: true
    root: msh/EU_868
    tlsEnabled: true
    username: test
  neighborInfo:
    updateInterval: 900
  serial:
    enabled: true
  telemetry:
    deviceUpdateInterval: 900
    environmentMeasurementEnabled: true
    environmentScreenEnabled: true
    environmentUpdateInterval: 900
owner: Test
owner_short: Test
```

garthvh, 2024-06-29:
code is here, meshtastic uses mqtt 3.1

https://github.com/meshtastic/Meshtastic-Apple/blob/62f93ee555b42b2715229e647d40773774d06589/Meshtastic/Helpers/Mqtt/MqttClientProxyManager.swift#L56

closed 2024-06-29 by garthvh

ufergus, 2024-08-30:
I am also experiencing this issue and I am confused by the closed as 'not planned'.  Are you stating this is a CocoaMQTT issue instead?

mkgin, 2024-08-30:
> I am also experiencing this issue and I am confused by the closed as 'not planned'. Are you stating this is a CocoaMQTT issue instead?

Let's see... 

I have mosquitto mqtt server with let's encrypt certs and tried to get the iOS app to connect with TLS.  I haven't had time to make more detailed tests, collecting logs on both ends. The iOS app works fine unencrypted (port 1833) but not TLS on 8883  valid host certificate and `require_certificate false` in the x509 listeners configuration.

The meshtastic public server with will known credentials does not run TLS so I would guess this feature is less tested and/or not confirmed to be working at all with mosquitto.

ufergus, 2024-09-01:
I am curious what MQTT brokers the iOS App can successfully connect to with TLS.  I have the Android Meshtastic app and the meshtastic firmware both negotiating successfully with the mosquitto.  So there is something unique about the iOS app, or the underlying CocoaMQTT or iOS libraries.  

As for the public meshtastic server, if the credentials are well known and included with the product, turning on TLS is just going to burn cpu cycles.

mmmorks, 2024-09-24:
I've also run into this with certificates that validate just fine when connecting with non-iOS clients. Seems like this is an issue in CocoaMQTT: https://github.com/emqx/CocoaMQTT/issues/448