meshtastic/Meshtastic-Apple #639: 🐞 [Bug]: Mesh Map Crashes

gfalgiano, 2024-05-09:
### Firmware Version

2.3.7 and 2.3.8

### What did you do?

When I view the mesh map and zoom in just a bit, the app crashes. Repeatable 100% of the time. I have 168 nodes reported in the app. Connected to a T-echo. 

### Expected Behavior

Should be able to zoom on map to see node locations. 

### Current Behavior

App crashes when zooming. 100% repeatable. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Mesh is hitting critical mass in my area! Great job Meshtastic team!

hutchisr, 2024-05-13:
Is this on iOS? I have had the same problem for several months, caused by the app using more than the allowed memory and being force-closed by iOS.

gfalgiano, 2024-05-13:
Yes iOS

Resolves if you clear the node database. Issue might be with having too many nodes?

Yom73, 2024-05-26:
A quick video to illustrate the crashing issue I am facing…
https://youtu.be/d-ep7OaXV_8?si=SJfs4fNp9QSPYFof

garthvh, 2024-05-29:
fixed in the App Store