meshtastic/Meshtastic-Apple #643: 🐞 [Bug]: GPS probably not sent to device and Mesh Map Issue in iPhone APP

ThomasF98, 2024-05-09:
### Firmware Version

2.3.8

### What did you do?

There seems to be an issue with new Nodes that get their GPS from the iPhone not showing up in the iPhone's 'Mesh Map' tab in the iPhone APP. I'm not sure if it is a problem with the node firmware or the iPhone App. This problem seems to be related to nodes that get their GPS location from the iPhone APP, because the one node that I have with on-board GPS (a LilyGo t-beam) doesn't seem to have this problem, but my Rak Wireless and Heltec V3 nodes do have the problem described below. Several weeks ago, I took some nodes off the air and replaced them with other nodes that I wanted to compare to the original nodes (different manufacture of the node hardware; RakWireless VS Heltec VS LilyGo). I discovered that the new nodes didn't seem to be 'seeing' nodes that the previous nodes were seeing on my iPhone's Mesh Map tab. I initially believed the issue was that the radios in the new hardware just weren't performing as well as the previous hardware that I had been using. However, recently I discovered that on my other existing nodes that were on the air continuously, that if I cleared the list of nodes in the iPhone APP 'Device' tab by performing a 'Reset NodeDB', that the existing nodes also no longer pick up the broadcasts from other nodes, 'learning' the new nodes locations and showing those nodes on the iPhone Mesh Map tab. What I think is happening is that for nodes that are on the air and have previously discovered other nodes and logged their GPS locations, the other nodes show up on the 'Mesh Map' because the node remembers the location of other nodes once it learns them. However, nodes are no longer 'learning' the location of new nodes once I reset the NodeDB. I suspect that it is because the new nodes aren't successfully retrieving their GPS location from the iPhone APP, and thus no longer propagate their GPS location to the mesh. Can other people with iPhones try going into the Device tab and clicking on 'Reset NodeDB'? Then, wait an hour or so and check the following; 1) How many of the previously known nodes showed back up on your iPhone 'Mesh Map' tab within a few hours? 2) Does your iPhone APP even know the location of your own node if you click on your own node's name in the 'Nodes' tab or is the "Node Map' button greyed out? 3) Does your own node show up in the map in the iPhone's 'Mesh Map' tab? 4) If you click on the name of each other node in your iPhone's 'Nodes' tab, for how many of them is the 'Node Map' tab also greyed out?

### Expected Behavior

Hopefully someone that understands the mechanism the nodes get their location from the iPhone APP can verify my observations and help get this issue resolved.

### Current Behavior

Once  you reset  your NodeDB, nodes no longer seem to know their location and no longer propagate their location to the mesh.

### Participation

- [X] I am willing to submit a pull request for this issue.

### Additional comments

Not sure if maybe this is the resurfacing of an issue that was addressed previously from this message thread: https://github.com/meshtastic/Meshtastic-Apple/issues/541

noahwilde, 2024-05-10:
Same issue here. iOS app version 2.3.8, firmware 2.3.5 on a Heltec v3. Just getting started with Meshtastic and was trying to figure out how to get updated gps from my phone to the node. Setting a fixed gps location works but no combination of settings seems to let it update automatically. Testing with an old android phone shows that it works. Reset node db and without adding a fixed location back in, no location is registered. Definitely seems like iOS is unhappy passing gps to the node automatically.

closed 2024-05-12 by ThomasF98

Yoshib8ec, 2025-03-19:
I have this same issue. I didn’t reset my node db though. I just rotated a node (node 2) on the same device to get it set up. When I put the original node (node 1) back on the device is when I couldn’t see my own node on its own UI mesh map. I’ve tried what feels like almost everything. I rotate it to an android and it works. My node (node 2) sync’d to the android device can see its own self and the other node on its mesh map though. My node (node 1) can see node 2, but not its own self. “Node Map” is greyed out as well.

Heltec v3
Firmware 2.5.14

ThomasF98, 2025-03-19:
Yoshib8ec: For me, this issue has been going on for a year and little is changed about the problem. For example, last week, I was logged onto a node that I have in a remote location. The node shows on the node map as being located at my home even though I was a many miles away and the node hadn't been at my home for months. I tried everything I could to go in and reset the node database, set the position to a fixed location and save, clear the fixed location so it would pull GPS location live from the Meshtastic app, tried changing it to use Device GPS position 'Disabled' and tried setting Fixed Position, clearing Fixed Position, changing Device GPS position to 'Not Present' and again trying to set a Fixed Position and tried clearing the Fixed Position. Never could get it to show the node on the map where I was at that time. I gave up hope long ago that devices without GPS could reliably obtain their location from the Apple Meshtastic app. It's one of those situations where sometimes you get lucky and the device successfully obtains the GPS position from the Apple Meshtastic App, and other times that there is nothing that I can do to get the device to pull the GPS location from the Apple Meshtastic App.

ThomasF98, 2025-03-19:
Yoshib8ec: As a follow up to what is discussed about this problem in this message thread, see the Github link below where I originally reported the problem on Github and was told the problem was 'fixed'. I don't know if the problem was actually fixed at that time and later broke again, or if it was never actually fixed, but at the time they told me the problem was fixed, it did appear to work so that problem ticket was 'closed'.

https://github.com/meshtastic/Meshtastic-Apple/issues/541

Yoshib8ec, 2025-03-19:
ThomasF98: I may have got it working. It’s showing on my map now, and hasn’t been intermittent for 20-30 minutes. I did do handfuls of things, so I’m not sure what exactly worked, but I’ll start with the very last thing I did. On the “Bluetooth” tab, press and hold the “connected” radio until there is a drop down pop up. Toggle through “mesh live activity” from the “play symbol” to the “stop symbol” and back. Once I did that it started showing on its own mesh map. Before that, I went into my phones Bluetooth settings and “forgot” all the other nodes my phone connected to in the phone itself (my 3 nodes, not the node db in the mesh app), and only left the one recognized in that setting. Those are the two things different that I had never done before. I did run through the other steps as well like resetting position, and just toggling on and off those settings. However, when I messed with the “mesh live activity” setting on the node my phone was connected to is when it started populating on its own map.