meshtastic/Meshtastic-Apple #658: 🚀 [Feature Request]: Show date and time in message list

PaulOckenden, 2024-06-03:
### OS

iOS, iPadOS, macOS

### Description

It would be lovely to see dates and times available for messages. RIght now I return to my Mac and see four new messages but I've no idea whether they are minutes, hours, or days old. I think there's space to append it to the small text lone name and id line. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-06-05:
There are time details in the long press menu on each message, and in the last message details on the message lists

PaulOckenden, 2024-06-05:
Sorry Garth - my humble apologies. I’d completely forgotten about the long press facility in the message list. What an idiot!

closed 2024-06-05 by PaulOckenden