meshtastic/Meshtastic-Apple #666: 🐞 [Bug]: Altitude in metres is reported as yards

clwgh, 2024-06-05:
### Firmware Version

2.3.11.2740a56 alpha

### What did you do?

Meshtastic iOS app `2.3.10 (956)`
iOS `17.5.1`

Fixed position stored on node. Altitude is in metres.

```
$ meshtastic --setlat xx.xxxx --setlon -x.xxxx --setalt 30
Connected to radio
Fixing altitude at 30 meters
Fixing latitude at xx.xxxx degrees
Fixing longitude at -x.xxxx degrees
Setting device position and enabling fixed position setting
$
```

Paired to the node via bluetooth in the app. Tap the <kbd>Mesh Map</kbd> icon, find the node on the map and tap it. Info box slides up and contains the altitude value but reports it as "yards" instead of "metres".

[image]

### Expected Behavior

Altitude value reported as metres since it is in metres.

### Current Behavior

Altitude has the correct value but incorrect unit text.

### Participation

- [ ] I am willing to submit a pull request for this issue.

clwgh, 2024-06-05:
I suspect there's more to it than just finding and replacing the string or else I would be happy to submit a PR with the mod.

garthvh, 2024-06-05:
All measurements in iOS are handled by your language and region settings

garthvh, 2024-06-05:
[image]

closed 2024-06-05 by garthvh

clwgh, 2024-06-05:
I don't care whether the units are metres or yards. The bug is that the app is taking a value in metres and reporting it as yards.

It should either report the set value as metres, or it should convert it to yards and report that value instead.

30 metres = 32.8 yards

clwgh, 2024-06-05:
I suspect you've coded it based on reading a GPS value and displaying in the OS units, so no conversion necessary since the GPS has the altitude. But you've not considered a fixed position where the value is always in metres, and so you end up displaying that either as 30 metres or 30 yards, the latter being incorrect because you haven't converted it from metric to imperial.

garthvh, 2024-06-05:
Everything converts from meters to the local units

clwgh, 2024-06-05:
Evidently not. 30 metres is not 30 yards, is it? The app is not converting it at all, simply reading and displaying the stored values and tagging on the wrong units.

garthvh, 2024-06-05:
Run meshtastic --info and look at the altitude on postion after setting it to 30meters

[image]

The app expects to get meters.

clwgh, 2024-06-05:
The lat, lon and alt don't appear in `--info`.

```
...
  "position": {
    "positionBroadcastSecs": 900,
    "fixedPosition": true,
    "gpsUpdateInterval": 120,
    "positionFlags": 11,
    "broadcastSmartMinimumDistance": 100,
    "broadcastSmartMinimumIntervalSecs": 30,
    "gpsMode": "NOT_PRESENT",
    "positionBroadcastSmartEnabled": false,
    "gpsEnabled": false,
    "gpsAttemptTime": 0,
    "rxGpio": 0,
    "txGpio": 0,
    "gpsEnGpio": 0
  },
...
```

The alt is set to 30, as reported in the opening post of this bug report.

```
$ meshtastic --setlat xx.xxxx --setlon -x.xxxx --setalt 30
Connected to radio
Fixing altitude at 30 meters
Fixing latitude at xx.xxxx degrees
Fixing longitude at -x.xxxx degrees
Setting device position and enabling fixed position setting
$
```

It appears correctly in `--export-config`

```
...
location:
  alt: 30
  lat: xx.xxxx
  lon: -x.xxxx
...
```

clwgh, 2024-06-07:
I did some more testing today.

Instead of configuring a fixed position I used the phone's GPS for position, and at this location the altitude was stored on the node as 26, ie 26 metres AMSL. The app showed this as "30 yards".

So at least this time there is a difference between the number stored on the node and the number displayed in the app, but it is still not a correct conversion because 26 metres = 28.43 yards, and 30 yards is 27.43 metres, so one of them is wrong.

Like last time the altitude on the node was not present in `--info` and was present in `--export-config`, with the only difference being how it got there. In today's test the value of 26 got there via the app from the phone's GPS. In the original report the value of 30 got there from me putting it there via the CLI.

location:
  alt: 26
  lat: xx.xxxxxxx
  lon: -x.xxxxxxx

This made me wonder – is the app simply reading the value, converting to yards and rounding to the nearest 10 or something? Then I check all the other nodes on the map and they are all multiples of 10 yards.

If I go to the node itself and look at the position log, I see, eg, 85ft, which is 25.9m which ties in with the 26m (0dp) shown in the CLI. The map node info when tapping the node shows "30 yards". 85ft is 28.3 yard, so it looks like it's rounded it up.

**Is the app simply rounding to the nearest 10 yards when displaying node info when tapping a node on the map?**

clwgh, 2024-06-08:
# Summary

For anyone finding this in future. The node info on the <kbd>Mesh Map</kbd> page, which is accessed by tapping a node on the map, contains information about the node. If the node is reporting altitude this is shown, **however it is rounded to a multiple of 10 yards** and cannot be relied on as an altitude measurement.

This behaviour is the cause of the apparent bug observed. Correct altitude here is 30 metres, as entered into the node. This is read by the app and converted to yards (32.8 yards) and then rounded to 30 yards to be displayed on the info panel. Thus making it appear that the value in metres iss being read and displayed as yards. Only after some additional testing did I discover this rounding behaviour.

To get an accurate altitude reading, go instead to the <kbd>Nodes</kbd> tab, find the node, tap it and select `Position Log` and look for the `Alt` column where you will find the most recent value as reported and not rounded.