meshtastic/Meshtastic-Apple #679: [Feature Request] support connecting to multiple nodes from the iOS app

72A12F4E, 2024-06-08:
### OS

iOS, iPadOS, macOS

### Description

https://github.com/meshtastic/Meshtastic-Android/pull/1078

Do this for the Apple apps

### Participation

- [X] I am willing to submit a pull request for this issue.

pqhf5kd, 2024-06-08:
Could the data for each node be shared between client devices over iCloud? 

Use case: connecting to portable node from macOS at one location. Later connecting the same node to iPhone, retaining nodeDB, metrics and messages.

72A12F4E, 2024-06-08:
> Could the data for each node be shared between client devices over iCloud?
> 
> Use case: connecting to portable node from macOS at one location. Later connecting the same node to iPhone, retaining nodeDB, metrics and messages.

This should be possible via CloudKit, but we might want to restrict certain data from syncing depending on how that might impact the UX. Would need to think about that more.

garthvh, 2024-06-08:
A backup and restore of the data from a previously connected node could work, displaying multiple users at once from changing perspectives is not a direction I want to go.

pqhf5kd, 2024-06-08:
@garthvh I'm not sure if you're responding to my suggestion, or @72A12F4E's original suggestion.

Please can you clarify 'displaying multiple users at once from changing perspectives'

garthvh, 2024-06-29:
The phone to user in meshtastic is 1:1, you can have different users with different channels and keys that display at the same time.  You don't log into what's app as two users at the same time.

closed 2024-06-29 by garthvh