meshtastic/Meshtastic-Apple #680: 🚀 [Feature Request]: Calculate and Display True Bearing of Nodes

pqhf5kd, 2024-06-08:
### OS

iOS, iPadOS, macOS

### Description

Calculate the true bearing (angle between yourself and other nodes, from North) and display in various views next to distance. 

The math is in the [Android code.](https://github.com/meshtastic/Meshtastic-Android/blob/5b3c78316b8accc789471f7ebdcd36e295c76682/app/src/main/java/com/geeksville/mesh/util/LocationUtils.kt#L217)

example: 
[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-06-29:
What is the real use case for true bearings? Is there a lot of surveying or meshtastic aircraft use?  Not sure it is really relevant for most users.

garthvh, 2024-08-04:
Closed by https://github.com/meshtastic/Meshtastic-Apple/pull/843

garthvh, 2024-08-04:
[image]

closed 2024-08-09 by garthvh