meshtastic/Meshtastic-Apple #681: Meshtastic app seems to still be using GPS even when no node is connected🐞 [Bug]:

therealchrisjc, 2024-06-08:
### Firmware Version

2.3.10.d19607b

### What did you do?

Enabled GPS in meshtastic app, always allowed access to use GPS

### Expected Behavior

GPS stops being used when there is no meshtastic node connected

### Current Behavior

Meshtastic sits in the background polling gps causing extra battery drain

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Keep up the good work, love it :D

garthvh, 2024-06-08:
The phone gps is used for a number of features, not just sending the location to the device, this is expected. You have granular control over what data is sent to the app so you can turn off the phone gps if you want.

closed 2024-06-08 by garthvh

therealchrisjc, 2024-06-10:
So using the GPS when its unnessercery is a feature?