meshtastic/Meshtastic-Apple #688: iOS 18 support

paoloponzano, 2024-06-11:
### Firmware Version

2.3.11

### What did you do?

iOS 18 dev beta 1 was released yesterday and I had a chance to test the Meshtastic app. I know it’s just the first beta but I think it would be good to keep track of the issues to solve them before stable release is published.

### Expected Behavior

Open the messages page, select any channel. 
The expected behavior is for the app to show the channel chat and text input box. 

### Current Behavior

After opening the channel, a blank page is shown. The chat won’t load even after waiting some minutes
[image]
[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

I’m not sure if I could take some logs without a Mac, but I could try in the coming weeks installing a VM with macOS to run xcode 
I will update this report if I find other issues related to the update

runestone74, 2024-06-11:
It seems like a new Bluetooth allow switch is needed in the app settings for ios to be able to connect. I got a connection at first but after a reboot the Meshtastic client seems to have lost BT privileges

garthvh, 2024-06-12:
Maybe in a month once there is a build, one day in is too soon

closed 2024-06-12 by garthvh