meshtastic/Meshtastic-Apple #692: 🐞 [Bug]: Disabling "Transmit Enabled" when setting Region upon fresh install doesn't disable TX

Hunter275, 2024-06-13:
### Firmware Version

2.3.12

### What did you do?

Upon a fresh install and connecting via Bluetooth for the first time you're asked to set the Region, set the Region and also turn off "Transmit Enabled"

### Expected Behavior

Region is set and TX is disabled

### Current Behavior

Region is set but TX does not disabled

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-06-13:
I can't replicate this, transmit is always enabled when setting region for the first time

closed 2024-06-13 by garthvh

Hunter275, 2024-06-13:
> Do you have a screenshot to what you are referring to? On freshly installed devices, until the region is set TX is disabled by default. The warning you get upon first connection is to set the LoRa Region. Are you saying TX disabled is staying disabled?

No, I want to set the region but disabled TX upon setup.