meshtastic/Meshtastic-Apple #696: Channels not displaying on iOS18

kalinchuk, 2024-06-15:
### Firmware Version

2.2.24

### What did you do?

Go to Messages > Channels > Primary Channel. The screen is blank. This happened occasionally on iOS17 but restarting the app seemed to fix it. On iOS18 there seems to be no workaround for it. 

### Expected Behavior

Shows messages for the channel. 

### Current Behavior

Not showing messages for the channel. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-06-15:
Ther is no iOS 18 support at all yet

closed 2024-06-15 by garthvh

kalinchuk, 2024-06-15:
So you close it hoping it will work on iOS18 when it's released to the public?

thebentern, 2024-06-15:
> So you close it hoping it will work on iOS18 when it's released to the public?

It's not something we're actively tracking right now. The 18 beta just got released at WWDC and is half baked at the moment. There is no reason to spend any time on this until the dust settles.