meshtastic/Meshtastic-Apple #711: 🐞 [Bug]: Messages aren't acknowledged in the message UI

72A12F4E, 2024-06-20:
### Firmware Version

2.3.11

### What did you do?

1. Launch App
2. Select Messages
3. Select Channels
4. Select Primary Channel
5. Type "Test" into the message field
6. Send message

### Expected Behavior

The message should send, and be acknowledged. 

### Current Behavior

The message is stuck in "Waiting to be acknowledged" state in the UI. If I kill the app and come back, it shows as "Acknowledged"

### Participation

- [ ] I am willing to submit a pull request for this issue.

GUVWAF, 2024-06-20:
I'm seeing the same on Testflight and an issue was already opened for this: #693

72A12F4E, 2024-06-20:
I tried this, left the app running for a while, and the last thing I saw from the device is this...

```
2024-06-20 10:16:42.526202-0500 Meshtastic[59747:29006439] [🗄️ Data] 💾 [Telemetry] Saved for Node: !33698B9C
2024-06-20 10:21:15.681635-0500 Meshtastic[59747:29006439] [🗄️ Data] 💾 Saved a new sent message from 862546288 to 0
2024-06-20 10:21:26.246538-0500 Meshtastic[59747:29006439] [🕸️ Mesh] 🚫 Don't Send an Empty Message
2024-06-20 10:21:31.714927-0500 Meshtastic[59747:29006439] [🕸️ Mesh] 🚫 Don't Send an Empty Message
2024-06-20 10:21:34.358364-0500 Meshtastic[59747:29006439] [🗄️ Data] 💾 Saved a new sent message from 862546288 to 0
2024-06-20 10:21:37.796252-0500 Meshtastic[59747:29006439] [🗄️ Data] 💾 Saved a new sent message from 862546288 to 0
```

closed 2024-06-23 by garthvh