meshtastic/Meshtastic-Apple #714: App shows Hourly Duty Cycle for licensed operators

nolith, 2024-06-23:
### Firmware Version

2.3.12

### What did you do?

Enabled hamradio licensed operator

[image]
[image]

### Expected Behavior

Hourly Duty Cycle warning should not be visible when configured as licensed operator.

### Participation

- [ ] I am willing to submit a pull request for this issue.

nolith, 2024-06-27:
Using the meshtastic CLI I realized that `lora.overrideDutyCycle` was still set to `false` after enabling the ham radio mode.

I set it to `true` using the CLI but in the iOS App i still see the duty cycle warning.

garthvh, 2024-06-28:
Closed by https://github.com/meshtastic/Meshtastic-Apple/pull/718

commit c8ddbd3b (2024-06-28): Merge pull request #718 from RCGV1/Duty-cycle-licensed

Fixes #714
closed 2024-06-28 by garthvh