meshtastic/Meshtastic-Apple #722: 🐞 [Bug]: Bluetooth no longer reconnects automatically after losing connection

tekstrand, 2024-06-26:
### Firmware Version

2.3.10

### What did you do?

1. Connect to a node
2. Move out of BLE range, causing a disconnect
3. Come back into BLE range
4. Note the device is shown in the Nodes list, shows a star for more recently connected node, but no longer automatically reconnects

### Expected Behavior

When coming back into range, the BLE should automatically re-pair/reconnect as it used to do.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-06-29:
Can you reconnect at all manually or do you have to forget and re-add the device?  What device?

garthvh, 2024-06-30:
I think this is a duplicate of this https://github.com/meshtastic/firmware/issues/4208 and probably firmware related as there are not BLE changes for quite a while.

closed 2024-06-30 by garthvh

tekstrand, 2024-07-15:
Sorry for the late reply, think my GH notifications are messed up. I am on the latest firmware (beta, 2.3.13) and the latest ios version. I just have to tap the device and it connects automatically, don't need to re-add. It even still has the star.

Device is the T-Echo.

If you think it's firmware I can open an issue there. @garthvh

tekstrand, 2024-07-16:
Alright so this is also happening with RAK4631's that I have