meshtastic/Meshtastic-Apple #724: 🐞 [Bug]: changing operator callsign may fail silently

nolith, 2024-06-27:
### Firmware Version

v2.3.13.83f5ba0

### What did you do?

I have a 433Mhz node with Call Sign set to IU5BON, after finding other hamradio nodes with a long name in the format CALLSIGN-ESSID I decided to change mine to IU5BON-9 to better identofy multiple nodes on the same callsign.

The app allowed me to save the configuration but nothing changed.

### Expected Behavior

The Call Sign, and node long name should change

### Current Behavior

I connected the serial port and this is what I found while sending my modified callsign

```
INFO | ??:??:?? 719 Telling client we have new packets 10
INFO | ??:??:?? 719 BLE notify fromNum
DEBUG | ??:??:?? 719 [Router] Add packet record (id=0x9a4fd6ba fr=0x00 to=0x54, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 priority=70)
DEBUG | ??:??:?? 719 [Router] handleReceived(REMOTE) (id=0x9a4fd6ba fr=0x00 to=0x54, WantAck=1, HopLim=0 Ch=0x0 Portnum=6 priority=70)
DEBUG | ??:??:?? 719 [Router] Module 'Admin' wantsPacket=1
INFO | ??:??:?? 719 [Router] Received Admin from=0x0, id=0x9a4fd6ba, portnum=6, payloadlen=26
ERROR | ??:??:?? 719 [Router] Can't decode protobuf reason='string overflow', pb_msgdesc 0x3c151aac
ERROR | ??:??:?? 719 [Router] Error decoding protobuf module!
ERROR | ??:??:?? 719 [Router] Can't decode protobuf reason='string overflow', pb_msgdesc 0x3c151aac
ERROR | ??:??:?? 719 [Router] Error decoding protobuf module!
DEBUG | ??:??:?? 719 [Router] Module 'Admin' considered
DEBUG | ??:??:?? 719 [Router] Module 'Admin' handled and skipped other processing
INFO | ??:??:?? 719 From Radio onread
INFO | ??:??:?? 719 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 719 encoding toPhone packet to phone variant=11, 12 bytes
INFO | ??:??:?? 719 From Radio onread
```

So I tested with the CLI

```
meshtastic --set-ham IU5BON-9
Connected to radio
Setting Ham ID to IU5BON-9 and turning off encryption
Writing modified channels to device
```

But the device didn't upgrade its config, with `meshtastic --info` I still get `Owner: IU5BON (5BON)`.

As a last attempt I did `meshtastic --set-owner IU5BON-9` that worked but removed my IsLicensed flag as the iOS App was no longer showing me as an ham. At this point I did `meshtastic --set-ham IU5BON-9` and now everything is ok.

Now the iOS app shows the right information but I am afraid of doing any config change from my phone as it may revert/not-work.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Overall I found very confusing that setting the ham radio mode is not consistent across all the clients.

As another datapoint, a friend of mine did the same stuff on Android and was able to set the nodename as CALLSIGN-15 but he eneded up with a default channel with encryption that was surprising as the default encription key was automatically removed from me using both iOS and the CLI.

nolith, 2024-06-27:
`todd-herbert` on discord pointed out to some relevant piece of code in the protobuf.

- https://github.com/meshtastic/protobufs/blob/a3030d5ff187091c9fbbd08dd797cca5085736fe/meshtastic/admin.options#L10C1-L11C37
- https://github.com/meshtastic/protobufs/pull/309/files

A few things:

- the UI shows a 8 bytes limit but includes the null terminator, this is not in line with the short name where the protobuf limit is 5 and the UI states 4.
- It is true that callsigns longer than 7 are very unlikely, but hamradio operators may have more than one device, other systems uses the `-ESSID` suffix (I think it's from X.25 specs) to identify different devices.
- When operating from another country (when allowed) we must prefix out callsign with a code that identifies the country we are in (i.e. a few months ago I was in Nevada and I had to ID as W7/IU5BON) this will not be possible with the current settings.

Can we relax a bit the protobuf requirements for the callsign? In the end it goes into the long name that is 39 bytes

garthvh, 2024-06-27:
The python api needs to be upgraded to set the ham parameters values properly.  There is some overhead on protobufs of a byte or two.

garthvh, 2024-06-27:
The app has been updated to only allow 6 characters, the size restriction is on the protobufs 8bytes is probably correct for 6 characters

closed 2024-06-27 by garthvh