meshtastic/Meshtastic-Apple #746: 🐞 [Bug]: Locally connected node reports Hop Count of 1

RPN707, 2024-07-02:
### Firmware Version

2.3.13

### What did you do?

Updated FW to 2.3.13. Allowed iOS to update app to 2.3.11 (1025)

### Expected Behavior

Would expect node connected to via Bluetooth to not report a hop count

### Current Behavior

iOS 17.5.1 - FW 2.3.13 - App 2.3.11 (1025)
Tried 3 nodes, T-Echo, RAK 4631, Heltec V3. All same behavior.
Uninstalled and reinstalled app.
Since updating app to 2.3.11, node shows "rabbit" hop count of 1 when connected via Bluetooth in the node list.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-07-16:
This data is provided by the node, I have not been able to replicate this with any recent firmware versions

RPN707, 2024-07-16:
Others on the (San Francisco) Bay Area Meshtastic Discord have confirmed
they're experiencing it as well. One guy even said he sees it with his TCP
directly-connected node. Some said they'll come post their details to the
bug board.