meshtastic/Meshtastic-Apple #750: 🐞 [Bug]: Bluetooth Connection

JonasJ333, 2024-07-04:
### Firmware Version

2.3.11

### What did you do?

Every time I have to connect to my Meshtastic devices I now have to forget the in Bluetooth and then reconnect them afterwards.

### Expected Behavior

I should be able to reconnect to a device that I had previously connected to. 

### Current Behavior

The connection either times out and gives the ten connection attempts or the connection screen just sits at the orange “communicating with device”

### Participation

- [ ] I am willing to submit a pull request for this issue.

JonasJ333, 2024-07-04:
I have also tried most of the basic troubleshooting items, turning Bluetooth off and back on, phone on airplane mode and back off, deleted the app and reinstalled, turning the phone off and back on.

garthvh, 2024-07-04:
This is a firmware issue

closed 2024-07-04 by garthvh

JonasJ333, 2024-07-04:
[image]

garthvh, 2024-07-04:
Upgrade your firmware

JonasJ333, 2024-07-04:
IOS App does not currently have an update in the App Store. The Meshtastic devices are all on 2.3.13 already.

JonasJ333, 2024-07-04:
The 2.3.11 I had referenced previously is the IOS app version.