meshtastic/Meshtastic-Apple #751: 🐞 [Bug]: device metrics export does not include voltage/current from power telemetry, nor does environment metrics export

akohlsmith, 2024-07-06:
### Firmware Version

2.3.10.956

### What did you do?

On OSX, the Environment Metrics tab displays power metrics too in the table, but when I export the CSV file the power metrics do not show up. Ok, fair but a little confusing, I'll export them from the Device Metrics where they are NOT shown, but the battery SoC and such are there. The current reported by power telemetry isn't in that export either.

### Expected Behavior

I would expect to have either a separate window/tab/etc. for power metrics and the ability to export them there, or have the power telemetry exported with the environmental metrics where they're being shown anyway. Ultimately, I'd expect what I see in the table to be in the CSV file.

### Current Behavior

I tried to export the environment metrics to a CSV file using the button provided so that I could graph the power telemetry I'm seeing in the table. Not finding the power telemetry in the CSV file, I tried to do the same from the Device Metrics tab, but they also do not show there.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

I am able to aid in debugging/testing, but do not have any experience building ios/osx apps. I'm a firmware dev.

garthvh, 2024-07-06:
Those power metrics should not be in the environement telemetry, I will get them removed from the table

akohlsmith, 2024-07-06:
Ok, but does that mean the power metrics will be included in the device metrics (the current is not there now) or an additional "tab" be created for power metrics where I can view/export this data?

garthvh, 2024-07-17:
Device metrics does not contain voltage, the power metrics needs some refactoring.  There is enough other environmental stuff to manage that I am not going to be blending the power data from environment metrics into there.

closed 2024-07-17 by garthvh