meshtastic/Meshtastic-Apple #752: 🐞 [Bug]: Channels URL for QR Codes Aren't Properly Formatted

AddisonTustin, 2024-07-06:
### Firmware Version

latest

### What did you do?

The syntax of a URI/URL requires that the fragment (the portion of the URI that follows a `#`) should be at the very end of the URI. When beginning to work on [meshtastic/Meshtastic-Android#953](https://github.com/meshtastic/Meshtastic-Android/issues/953) I noticed that since `?add=true` followed the fragment, Java's native URI feature does not properly parse `add=true` as a query param, and instead includes it as part of the fragment.

There's some cleanup I'd like to do as a part of [meshtastic/Meshtastic-Android#953](https://github.com/meshtastic/Meshtastic-Android/issues/953) to use the URI directly rather than doing manual string processing. So this becomes a small blocker.

### Expected Behavior

The query params (`?add=true`) should be included in the URI before the fragment.

### Participation

- [X] I am willing to submit a pull request for this issue.

### Additional comments

I'm willing to submit a PR, but I don't currently have a dev environment setup for iOS application development. So if there are others willing to tackle this before I have an opportunity to, that would be appreciated 🙂

garthvh, 2024-07-06:
It is after the hash so it does not get sent to the server. React uses this order as well.

AddisonTustin, 2024-07-06:
The actual channel configurations are in the fragment to [reduce the chance of leaking channel keys](https://github.com/meshtastic/Meshtastic-Android/issues/126), right? Not sure if it's as meaningful a distinction for the query string. But if it's not a super compelling reason, happy to close this issue and just process the QR code link appropriately in the Android client.

garthvh, 2024-07-08:
I moved it before the hash, I can handle both easily enough, probably a pain in js too

garthvh, 2024-07-10:
This is in app review

closed 2024-07-10 by garthvh