meshtastic/Meshtastic-Apple #756: 🚀 [Feature Request]: Sharing QR codes/links is confusing, potential for data loss

jasonfish, 2024-07-07:
### OS

iOS

### Description

It's nearly impossible to decipher how sharing a channel QR code going to work before sending, or before accepting. I created four links/QRs from a known-good device with only Primary and Admin, then factory-reset another device to detail what happens. I've found that it doesn't appear to be possible to Add channels to another device, and that the recipient could unintentionally yet easily be sent and use a Replace All link.

First, when creating the QR in the iOS app, the button text can read either "Replace Channels" or "Add Channels." This is confusing; it's not clear if the button text is the current state, or is the state that it _will change to_ by tapping it. Generally the text on a button indicates what _will_ happen _if_ you tap it. **I suggest two radio buttons to clearly select one or the other.**

In any case, here's what I found.

If the button text reads Add when the QR/Link is created, with only admin selected
Tapping the link on the receiving device pops a “Replace all channels?” prompt
Accepting this prompt fails with a “channel names must be unique” error

If button reads Add, with both primary and admin selected
Tapping link pops same “Replace all channels?” prompt
Fails with same “channel names must be unique”

If button reads Replace, only admin selected
Tapping link pops same “Replace all channels?” prompt
Succeeds by replacing all channels with admin only (this is the only scenario that definitely works as expected)

If button reads Replace, both selected
Tapping link pops same “Replace all channels?” prompt
Succeeds by replacing all channels with both (this may also work as expected, or maybe it only added admin? no way to tell)

The sharing page needs a UI rework, IMO. It's WAY too easy to send someone a link to Add a channel, only to end up with them replacing all of their channels with just that one. And the link-tapper should be told whether they're about to Add or Replace their channels, do not reuse the same dialog language.

[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

jasonfish, 2024-07-07:
I visited a remote node today to add my new admin channel. I chose to Replace channels when creating the link. When tapping the link on the remote node, I got the "Replace All?" prompt and accepted. That node now has a Primary and _two_ admin channels. And due to /Meshtastic-Apple/issues/743, I was unable to delete the old one.

Add vs Replace needs help. :) I'm absolutely willing and able to do a screenshare session to work on the issue together, do some real-time building and testing to find a good solution.

garthvh, 2024-07-07:
I think you have a lot of variables going on, I just saved channels as normal from a number of QR codes.

jasonfish, 2024-07-08:
Not sure what extra variables I may have. If I try to add channels to a device, it asks if I want to replace them, then fails to do either. When I try to replace channels on a device, it adds the ones I selected, but keeps one that already existed. I've gotten feedback from others who also can't figure out how this is supposed to work.

garthvh, 2024-07-09:
I can't replicate this behavior, there are new debug logs in the 2.3.15 firmware that may help you see what is happening in the device and app at the same time.

closed 2024-07-09 by garthvh

jasonfish, 2024-07-09:
Can you not replicate the reused dialog header on the recipient's device? No matter if you choose Add or Replace when creating the link, when tapped it always asks "Replace all channels?" Fixing just this could help users know what's about to happen.

jasonfish, 2024-07-10:
Using the same links I generated for my initial tests, the latest release from the app store now repurposes the Add language on the dialog, instead of Replace, no matter which link type is tapped. So it's reversed from before. :)