meshtastic/Meshtastic-Apple #760: 🐞 [Bug]: Direct Messages go to channel 0 by default

RustMouse, 2024-07-08:
### Firmware Version

2.2.24

### What did you do?

I changed the default channel to encrypt my location, and set the default settings (Long Fast) in channel 1.  
[image]
[image]

### Expected Behavior

I should be able to message in the channel, identify local nodes, and send direct messages to them. The messages should go out on the channel that matches the node I'm trying to reach
[image]

### Current Behavior

I can broadcast messages in the public channel, but the direct messages to known nodes fail.  When I look at the node list, the node appears on channel 1,  but when I message directly, it gives me "No Channel" error.  I've tried with two different nodes.  I can receive the messages on my device, but I cannot respond to it.  
When I look at the failed message detail, it appears to have tried to send out on channel 0, regardless of the channel the node is operating on.

[image]
[image]
[image]
[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-07-08:
Direct messages go over the channel they share with the node in question, it is managed by the firmware

closed 2024-07-08 by garthvh

RustMouse, 2024-07-08:
Strange - without changing anything, I can communicate with VXD3, but only after they messaged me again, though the message status still shows channel 0.
I tried the same with B2B9, and still have a "no channel" error

GUVWAF, 2024-07-08:
The app will always leave it at 0 and then the firmware fills it in. 
It can take a while before it updates. You can delete the node from the list to speed it up, because it will request a new NodeInfo then.

RustMouse, 2024-07-08:
Thank you,
Do you know how frequently this will update if the node is left in the list?

GUVWAF, 2024-07-08:
By default a node sends out its NodeInfo every 3 hours.