meshtastic/Meshtastic-Apple #766: 🐞 [Bug]: channels disappers from Messages -> channels

nolith, 2024-07-09:
### Firmware Version

2.3.13

### What did you do?

I have 3 channels with index 0, 1, and 2.

When I open the Messages->Channels tab, sometimes I do see a subset of channels.

### Expected Behavior

I expect to see always all my channale

### Current Behavior

Today, for example, channel 0 was not in the list. I had to force-close the app ands reopen to make it show up again.

A few hours ago the channel list was behaving as expected.

### Participation

- [ ] I am willing to submit a pull request for this issue.

RPN707, 2024-07-10:
I've noticed this too. I close the app completely and reopen it. The channel tab usually reappears. I've also noticed that, when this happens, one or more of the Settings, Channels config options is also missing.

tekstrand, 2024-08-21:
I've also noticed this one. Particularly when disconnecting from a node, connecting to a different one, and then reconnecting to the original.

closed 2024-10-20 by garthvh