meshtastic/Meshtastic-Apple #789: 🐞 [Bug]: Node List - Long press - Delete Broken

RPN707, 2024-07-12:
### Firmware Version

1.3.16 (1069)

### What did you do?

In the node list, a long press on a node brings up a "context" menu. There are now 3 items. The delete item does not delete the node from the node list.

### Expected Behavior

Node is not deleted. It remains in the list.

### Current Behavior

Choosing Delete returns to the node list with the selected node still visible. The new Delete item in the Node Detail DOES correctly delete the entry

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

It was nice having Exchange and Trace directly from the node list. Couldn't they be both places?

garthvh, 2024-08-03:
Closed by https://github.com/meshtastic/Meshtastic-Apple/pull/835

garthvh, 2024-08-04:
Fixed in #2.4.42

closed 2024-08-04 by garthvh