meshtastic/Meshtastic-Apple #792: 🐞 [Bug]: Channels Chat Broke

kjake, 2024-07-15:
### Firmware Version

2.3.17

### What did you do?

Updated iOS app to 2.3.17

### Expected Behavior

Channel messaging should be available and work like the previous revision

### Current Behavior

Since the iOS app updated to 2.3.17, there is no longer a way to send messages to channels - tapping on a channel just results in the blank screen.  Direct Message still works.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-07-16:
Is this iOS18?

kjake, 2024-07-16:
> Is this iOS18? 

Turns out it is! How embarrassing that I forgot I was running the beta already. So this isn't a high priority thing then, but know that it was working as expected on iOS 18 this morning before the latest version of the iOS app came down from the AppStore.

garthvh, 2024-07-16:
It was working for me with beta 3, but the early betas are broken again. Seems kind of inconsistent. This build is from Xcode 15.4 so I can't check for 18

closed 2024-07-16 by garthvh