meshtastic/Meshtastic-Apple #799: 🚀 [Feature Request]: Filter out MQTT Nodes you can't send DM's to

garthvh, 2024-07-17:
### OS

iOS, iPadOS, macOS

### Description

Don't show contacts that are multiple hops away on the default Channel via MQTT as you can't really message them anymore

### Participation

- [ ] I am willing to submit a pull request for this issue.

todd2982, 2025-01-13:
How do you see this impacting users connected to a private MQTT server that does not get forced to the zero hop count?

violarulan, 2025-04-21:
I think it's only should be applied to nodes that are discovered via public MQTT. With a private broker, you still can DM them hops away.

hayschan, 2025-04-28:
This is affecting private MQTT server as well, which should not be the intended target. 

It should only target the default public MQTT server only.