meshtastic/Meshtastic-Apple #802: 🐞 [Bug]: when power telemetry is enabled, ios/osx app saves "all zeroes" device metrics entry for received power telemetry packets

akohlsmith, 2024-07-17:
### Firmware Version

2.3.16.1069

### What did you do?

Sonoma 14.5 mac connected to t-echo over bluetooth. Monitoring device metrics of a mesh-connected WSL. When the WSL has power telemetry enabled, the osx app shows many "all zero" (time: 0, channel utilization: 0, airtime: 0, voltage: 0, SoC: 0) telemetry rows and the graphs are very "sawtooth"-y because of the errant "all zero" entries interspersed amongst the valid device metrics data.

If I turn off power metrics on the WSL, the data and graph shown in the osx app is good. If I turn on power metrics on the WSL, the "all zeroes" data starts getting recorded again. 100% reproducible.

### Expected Behavior

Power telemetry packets should be ignored and not misinterpreted and stored in the database. Ideally, another "tab" for "power telemetry" should be available where the voltage and current is shown in the table and graphed, and allowed to be exported via CSV.

### Current Behavior

When the app receives any telemetry packet, it tries to extract the data based on the telemetry variant. See `MeshPackets.swift` starting around line 652.  Line 660 and 699 attempt to extract data from device metrics and environmental telemetry variants, respectively, but unfortunately if a telemetry variant does not match, lines 689-692 still execute but since nothing was extracted, we get a row of "all zero" data added to the node's telemetry list.

Ideally, a row should only be added to the telemetry data if there was data successfully extracted from the protobuf.

Thanks to @mian on Discord for helping me "see through" the Swift code; I'm not a Swift developer.

### Participation

- [X] I am willing to submit a pull request for this issue.

ianmcorvidae, 2024-07-17:
Specifically, `telemetry.metricsType` is set to 0 for device metrics and 1 for environment metrics, but it would already have been 0 as of initialization on line 652.

Potential fixes based on my nearly nonexistent Swift/CoreData/etc. knowledge:

* change device to 1 and environment to 2, to let 0 for that actually be a proper "unknown" value
* start using -1 or something as unknown, add a fallthrough else to the conditional to set that
* (possibly including one or both of the above) implement cases for all the telemetry variants, even if they do nothing but set that `metricsType` value

(I'm mian over on discord, different name here)

garthvh, 2024-07-17:
This is fixed in test flight

closed 2024-07-17 by garthvh