meshtastic/Meshtastic-Apple #803: Implement exporting range test csv files

RubyMetro, 2024-07-17:
### OS

iOS, iPadOS, macOS

### Description

While android possesses the ability to export CSV files for the range testing module, iOS does not, despite having the ability to turn the range testing module on to any radio connected to it. I am requesting that an option be added to the range test module page in radio configuration to export a CSV file containing range test data to the onboard files app.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

A proposed solution was to "just use the position logs" but this defeats the purpose of the feature being available in the firmware. Functionality should be the same between all platforms.

garthvh, 2024-07-17:
Android also does not have the ability to download the actual range test csv file, it builds one from the 500 packet log.

RubyMetro, 2024-07-17:
> Android also does not have the ability to download the actual range test csv file, it builds one from the 500 packet log.

That is correct. I am talking about exporting the log to a CSV file like the android app is able to for easy use with Google Maps.

closed 2025-07-18 by garthvh